realsenseai/librealsense #6801: Connectors needed to sync multiple cameras

SaraGagliolo, 2020-07-10:
Hi,
I'm a beginner in sync purposes. I have a question about how to realize the material connection between the Intel RealSense D435i and a breadboard, hosting the connections to sync it with two other different cameras (Flir Blackfly S).

From the D435i point of view, a whitepaper (https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras) suggests to use the components ASSHSSH28K152 and SHR-09V-S-B, but I suspect that the indicated ASSHSSH28K152 jumper does not fit with a breadboard. In a scheme of the same whitepaper the component SM09B-SRSS-TB is reported: is it suitable to receive the terminal of the ASSHSSH28K152 jumper adapting it to a pin terminal? If yes, can I use male/female traditional jumper wires to link just the full positions of the SM09B-SRSS-TB connector to the breadboard (10 and 11 in the scheme of the whitepaper)?

I hope the question is reasonable and clear. Thanks in advance for the support.
BR,
Sara

RealSenseCustomerSupport, 2020-07-10:
----------------------------------------------

Hi

The ASSHSSH28K152 and SHR-09V-S-B are the assembly parts for you to build the sync cable Plug for connecting to D435i AUX port.
The SM09B-SRSS-TB is the mating part (receptacle) for the Plug. It is same one as what you see on the AUX port of D435i.

Please note the tricky part for the sync feature of D435i - it is somewhat picky on the trigger frequency while it is configured as a Slave cam. So I suggest you always to make D435i be the Master cam, and make your "Flir Blackfly S" be the Slave cam.

Hope I answered your question.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

SaraGagliolo, 2020-07-11:
Very helpful, thanks. Referring to the whitepaper I cited before, can I use the same SM09B-SRSS-TB that you clarified to realize the connection with the breadboard? Something similar seems linked with the signal generator in the picture.
Thanks for your interest.
BR
Sara

RealSenseCustomerSupport, 2020-07-11:
----------------------------------------------

Regarding the receptacle of SM09B-SRSS-TB, please note that its pitch size is 1.00mm, but most breadboard is designed for 2.54mm spacing. So I think you would need to do some minor soldering work - just solder the two wires (see the White Paper, the Figure 3 on page 4) on a header (2.54mm) and connect it to your breadboard.

Thanks

closed 2020-07-25 by RealSenseCustomerSupport