realsenseai/librealsense #6802: How can I save .ply file with both rgb color & vertex normals at the same time?

usarawgi911, 2020-07-10:
As described in https://github.com/IntelRealSense/librealsense/issues/6194 we found

1. ply with vertex normals, not with rgb color (like in export_to_ply.py)
2. ply with color, not with vertex normal (like the code below)

frames = pipe.wait_for_frames()
aligned_frames = align.process(frames)
aligned_depth_frame = aligned_frames.get_depth_frame()
color_frame = aligned_frames.get_color_frame()
pc.map_to(color_frame)
points = pc.calculate(aligned_depth_frame)
print("Saving to 1.ply...")
points.export_to_ply("1.ply", color_frame)
print("Done")

How can we save .ply file with both rgb color and vertex normals at the same time?
Thank you, really appreciate any help!

MartyG-RealSense, 2020-07-10:
Hi @usarawgi911   There was a RealSense user who created a C++ script and PCL to save a ply with voxels instead of vertexes.  Would this meet your needs, please?

https://github.com/IntelRealSense/librealsense/issues/6411

Voxel graphics will typically have more 'boxy' points.

[image]

MartyG-RealSense, 2020-07-26:
Hi @usarawgi911  Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-08-08 by MartyG-RealSense