realsenseai/librealsense #6804: Resonating lines

bobdavies2000, 2020-07-11:
I am using the L515 on the latest Windows 10.  There was a new librealsense released today and I will try that next but I have often come across images like the one below where curved lines migrate across the image.

[image]

The lines migrate from right to left and are often not present - especially after dark.  The environment is indoors but there are windows nearby and indirect daylight.  The lines are also in the depth data as well so this is not just an artifact in the RGB Depth image.  Have others seen this?  Is there anything I should try to remove these lines?  They feel like they are resonance in the scan of the L515.

Bob Davies

dorodnic, 2020-07-13:
Hello Bob,
It seems that you have additional ToF / Structured Light sensor running in background. 
We are going to publish multi-L515 hardware synchronisation guide very soon

bobdavies2000, 2020-07-13:
Thanks.  My test machine was running at the same time.  Now I know how to prevent this.

closed 2020-07-13 by bobdavies2000