realsenseai/librealsense #6805: D435 cameras not recognized

digitaldawg, 2020-07-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | unavailable at this time | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC|

### Issue Description

 I have 12 Windows10 machines with a D435 camera & Leap Motion attached.  Each time we boot up the group of computers 1-4 of them don't recognize the RealSense.  Which machine changes randomly on each boot up.  If we unplug-replug the USB it recognizes the camera.

I have a second setup having the same issue.  I have a Windows 10 machine with 2 D435 connected.  It randomly doesn't recognize one or both of the cameras.   The cameras are on 30' powered USB3 extensions.  I get a variety of errors - see attached images.

I'm constantly having to unplug-replug the USB cables.  Sometimes at the PC, other times at the camera.  I really need a solution!  I can't continue to play the unplug game.

MartyG helped on the other site.  I may try to use a powered USB3 hub.  
In the mean time, are there other options?

[image]
[image]
[image]

MartyG-RealSense, 2020-07-11:
Hi again @digitaldawg  You mentioned that you are using 30' powered USB3 extensions, which is new information .   If you are using USB cables with 'active repeaters' to extend the range of the cameras then this may have a bearing on your issues.  Whilst you can connect two or more together to create an extended length (such as 2x15 for a 30 ft length), in Intel's white paper document on multiple cameras they provide the following cautionary message about repeaters:

**********
Unfortunately the quality of these vary tremendously.  We have found that the following appears to work quite well.  We have successfully strung 3 of these together: https://www.centralcomputer.com/ 
***********

Alternatives to using repeater cables for long-distance wired camera connection of up to 100 meters include:

- Ethernet network cabling

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

-. Fiber optic cabling, such as that provided by Newnex.  They are a supplier of industrial grade equipment snd the officially recommended purchasing source for RealSense USB cables in the data sheet document for the 400 Series cameras.

https://m.youtube.com/watch?v=GLQgR1jT04M

digitaldawg, 2020-07-11:
Marty,

FYI - This is the extension cable we are using:
https://www.amazon.com/gp/product/B076C6PDNH/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 <https://www.amazon.com/gp/product/B076C6PDNH/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1>

The other challenge I have is that these computers are currently part of an exhibition in China.  All trouble shooting, and tests have to be done remotely.  I’m not able to quickly try new things.

The link to centralcomputer site lead to a 404.

Do you have any thoughts about the startup/powerup sequence of the cameras and Windows10; and how that relates to the OS recognizing them or not?

Thank you -
Scott Stevens
Interactive Specialist
Victory Hill Exhibitions
702-335-1465

MartyG-RealSense, 2020-07-12:
I have fixed the link, which was correct but did not launch from a GitHub page.

May I confirm please whether the camera detection problem occurs during the first boot-up of the PC from a shut down state?  It has been a known problem that the camera may work successfully on the first boot but then not be detected on the second boot after restarting ,without shutting down first and then pressing the power button to boot.

digitaldawg, 2020-07-13:
I problem occurs on startup after shutdown.

I'll check the firmware version tonight.

I'll let you know . . . .
Thanks for taking the time to help on this!

digitaldawg, 2020-07-14:
The day in China started out OK, then went to hell.
I’ve attached the RealSense Viewer log so you can see when I was trying to deal with USB2.1 situation below. ( have a newer log with more errors on the machine in china,  I’m trying to get the machine turned on so I can send you the longer log)  And some screenshots of errors.

When the computer started up we had the USB cables for both cameras connected.
They both were recognized.  I tested them in the Viewer, and both were showing depth feed.
I started my software, and it worked correctly using the depth from both cameras.

THEN, the local crew shut the computer down, and powered it up again.
I’m unaware how long between shutdown & startup, probably only a few minutes.

Then, neither camera was recognized.
I tried replug USBs at computer = nothing.
I tried restart computer = nothing
I had them replug the USBs at the cameras and the PC recognized them, 
	but one registered as only USB2.1 in the viewer.  
I had them replug that camera’s USB at computer = still USB2.1
	in this state my software can’t get depth feed.
I had to pause trouble shooting for an event there.

I’m truly baffled & frustrated about why would the PC recognize them the first time, but not the second.
I also don’t understand the difference between unplug/replug at the computer versus at the camera. There seems to be more of some kind of re-boot of the camera when unplugged at it.

In the attachment you can see the cameras have the current firmware.

Sorry to bother you about the link to the centralcomputer.com, but it only takes me to a generic search page on the site; not a product. (I used the link on GitHub)
Is the product a Unitek Y-3004 USB 3.0 16’ extension?

Thank you -
Scott Stevens
Interactive Specialist
Victory Hill Exhibitions
702-335-1465

MartyG-RealSense, 2020-07-14:
I think the Central Computer site was just being mentioned in general as a recommended purchasing source for extended cables, rather than referencing a specific product.

I cannot see an attachment unfortunately.

Due to the time-critical importance of your event, I will seek input from Intel about your problems.

digitaldawg, 2020-07-14:
I know you mentioned you couldn’t see the attachments, but in case you can forward them
I’ve attached 2 logs:
librealsenseRealSenseLeap = the computer that has a RealSense & LeapMotion connected
librealsense2RealSense = the computer with the 2 RealSense connected.

I was working on the computer with RealSense & LeapMotion connected.
I was trying to update the firmware.
As I had the RealSense Viewer active the viewer was showing the camera jumping between connected and not.  
At one point it showed me the pop-up for updating the firmware.  I clicked on it and it started to show the progress bar, Then it started the jumping between connected and not again.
This should be in the Log.

Please help!

MartyG-RealSense, 2020-07-14:
Please do not try to update the firmware right now, as you risk disabling the cameras if the update goes wrong due to an erratic connection.  I have sought internal assistance for you.

digitaldawg, 2020-07-14:
I’m afraid it may be too late for the camera that was going crazy during update.

MartyG-RealSense, 2020-07-14:
Okay, so it is just one camera so far that has broken?

MartyG-RealSense, 2020-07-14:
I have consulted with Intel.  Can you tell me please if Victory Hill has a Non Disclosure Agreement (NDA) with Intel, please?

digitaldawg, 2020-07-14:
No. We do not have a NDA with Intel. 

-Scott

MartyG-RealSense, 2020-07-14:
Ok, thank you very much.  

Do you have any official short camera cables at the exhibition site, please?

digitaldawg, 2020-07-15:
Yes. 
The computers, which also have the LeapMotion sensors, are using the factory cables to connect the RealSense cameras. 

The double camera setup has the cameras mounted over head. It would be a challenge to either unmount them; or move the computer close enough to use the original cables. 

-Scott

MartyG-RealSense, 2020-07-15:
I am back now.  If the affected camera is on the end of a 30' repeater cable and mounted overhead then it will be a challenge to recover it if you cannot disconnect it and connect the camera directly to a computer with the short cable.  The alternative would be to keep trying the recovery process through the 30' cabling and hope that (a) the camera can be detected, and (b) if it can be detected, the connection holds for long enough for the recovery process to complete an update.

If the RealSense SDK is installed on the Windows computers at the exhibition and it was installed using the Windows installer file from the SDK Releases page then you can access the **rs-fw-update** program through the Windows command prompt interface (go to the text box at the bottom of the Windows screen, next to the corner button where you shut down, and type command into the box to find the Command Prompt).

digitaldawg, 2020-07-15:
The camera I may have messed up is on a PC with the camera connected via the factory USB3 cable.(only, short distance)
There is a spare onsite. I’ll probably change it out.

I’m more worried about  the ongoing recognize issues.  
I’m going to try an experiment today. I’m going to have the local crew unplug the LeapMotion and restart the computer with just the RealSense connected and see if the PC recognizes it.
If I get a reliable connection, I’ll try to update the firmware.

I had some success with the PC w/ 2 cameras last night.
After a hard reboot of the cameras by unplugging USB at the camera; they both worked.  And more importantly they were both recognized after a Shutdown & Startup.
I’ll remain optimistic that those will remain OK since they have the newest firmware.

Another wrinkle in all of this - 
I’ve had these same PC setups here in our office.  For the 2 camera setup I never had any recognition problems here.  The only difference is that the computers in China are using 220V 50hz.  I wonder if the computer’s onboard power supply generates a slightly different voltage for the USB that the cameras are sensitive to, especially during startup.

I’ve ordered a USB3 voltage tester.  Unfortunately it may take 2 weeks until I can get it to the crew in China.

Any word from other Intel support?
Do I need to sign a NDA?

Thanks
Scott

MartyG-RealSense, 2020-07-16:
Regarding the request to apply for a RealSense NDA, I have passed your contact details from the comments above on to someone who can direct your enquiry towards people responsible for handling applications and they can take up the request with you.  In the meantime until an NDA can be successfully registered, I will be leading your support case.

Usually, the mains electricity power goes into the computer and is converted into voltage in the computer's power supply unit, and the USB ports draw from the power supply,  So that should compensate for local variations in power supply frequency wherever in the world you are. 

RealSense cameras are sensitive to the state of the USB port though.  This is why using a mains electricity powered USB 3 hub instead of attaching the cameras to the computer directly or through an unpowered ("passive") hub that draws its power from the computer PSU can help to resolve problems with camera connections that may occur if the computer's built-in USB ports have power supply instability.

Having said that, I do recall a case some years ago where a company was having RealSense stability problems and they thought that it may be related to fluctuations in their local mains electricity supp;y.  I believe that they used an uninterruptible power supply (UPS) pack to compensate for mains electricity supply spikes.

I had concerns about having a Leap Motion near the RealSense cameras.  If the Leap Motion is based on infrared, then RealSense 400 Series cameras can interfere with nearby non-RealSense infrared cameras (the 400 Series itself does not experience interference in return).  An example would be a RealSense camera interfering with a built-in IR camera in a laptop.  You have tested the project successfully in your office though, so you do not seem to be having such issues in your particular case.

MartyG-RealSense, 2020-07-16:
Can you provide your email address please so that your NDA enquiry can be processed by Intel, please?

digitaldawg, 2020-07-16:
This email is my personal.

My work email is:
Scott.stevens@vhexhibitions.com

MartyG-RealSense, 2020-07-16:
Thanks, I got it and have passed it onward.   you can edit it out now if you don't want it to be public.

digitaldawg, 2020-07-20:
Marty,

Please have someone contact me ASAP.
My troubles with the PC that has 2 cameras attached continue.

FYI - I updated the firmware on the cameras that are connected to the PC with the LeapMotions.  We're are going on 3 days without trouble.

Thank you -
Scott Stevens

MartyG-RealSense, 2020-07-20:
As you do not yet have an Intel NDA agreement, I remain the support team member handling your case.  Do your problems with the PC with 2 cameras attached remain the same as described previously or is there new information, please?

digitaldawg, 2020-07-20:
Marty,

Concerning the PC with 2 D435s:

After I updated the firmware the cameras were recognized error free for 3 days.
Then they started the errors.  It has been a variety of non-recognizition: sometime just 1, other times both.  Unplugging the USB cables at the computer doesn’t really help.
When they are not recognized the only solution seems to be unplugging the USBs at the cameras.

Over the weekend replugged at the cameras and was able to have computer to recognize them, but one of them only registered as 2:1 in the Viewer.  Numerous replugging did not change it.

With one of them at 2:1 it means I get no depth info from it.

I have numerous screen shots of the error pop-ups from the RealSense Viewer.  I also have the logs from it.  But since you said you can’t see attachments what do you suggest?

Thank you -
Scott Stevens
Interactive Specialist
Victory Hill Exhibitions
702-335-1465

MartyG-RealSense, 2020-07-20:
During this case you do not mention having tried the **Reset Hardware** option in the RealSense Viewer (which is under the **More** menu) to reset the camera without having to unplug-replug it when it is detected as USB 2.1.  Can you confirm whether you have tried this, please?

The computer with 2 cameras is the one that is using the 30' active repeater cables?

https://github.com/IntelRealSense/librealsense/issues/6805#issuecomment-658475168

digitaldawg, 2020-07-20:
I didn’t know about the Reset Hardware.  I’ll try it tonight.

Yes, this is the computer using the powered 30’ extensions.

-Scott

MartyG-RealSense, 2020-07-20:
I considered the possibility that cameras in general might have problems with a chain of active repeater cables and googled for **usb active repeater camera not detected**.   Here's an example of problems with 30' of repeater, like in your case:

https://community.spiceworks.com/topic/2184536-having-trouble-connecting-usb3-camera-cord-length-issues

One person tried connecting the repeater cabling to a powered USB 3 hub to increase the amount of power provided along the cable distance.

digitaldawg, 2020-07-20:
For reference:
I’m using 2 D435s in another exhibit.  They are each connected to an Intel NUC computer via the same 30’ extension I’m using now.  In this case we did not use the power option on the 30’ cable.  I’ve never had a problem with this setup.

If my problems were with an exhibit here in the states I’d try your suggestion of the fibre optics.  It is really difficult for me to get equipment into China.  It is also challenging sometimes to describe trouble shooting thru the Chinese - English language barrier.

Is there an ETA on the NDA?
Thanks
Scott

MartyG-RealSense, 2020-07-20:
I'm not involved with the NDA registration process, so have no way to know what stage an application is at.  I assume that your provided email address would be the method they would use to make first contact.

If you cannot change the cables, is there any way for the local exhibition crew to mount the NUC computer up at camera level and use the normal short cable?  That would have its own challenges such as power supply, but it would likely be far easier to get a 10 m mains power extension reel locally than new repeater cables.  You might also need a 10 m HDMI cable to reach from the high-mounted NUC to a monitor at ground level,   

OR take remote control of the elevated NUC from a PC on the ground so the NUC desktop is on the ground PC's screen.  That should be doable with purely software instead of hardware.

digitaldawg, 2020-07-21:
Tonight’s version:
At startup only one camera was recognized, at 2.1 USB
I tried the Hardware Reset numerous times = nothing.
Tried unplug-replug USB cables at computer = no change
Tried Hardware Rest for the 2.1 camera again = no change
Tried unplug-replug USB cables at each camera =
	both recognized, one still at 2.1
Tried Hardware Rest for the 2.1 camera again = no change

In terms of move PC and use original factory cables:
The cameras are mounted on small arms over an interactive table that is 3’ x12’.  The factory cables are not long enough to follow the mounting arms.  I would probably need 6’ cables.
Mounting the PC that high would take some custom fabrication and installation time.  Both difficult at this time.

I have the USB voltage tester, and will ship it to China tomorrow.

-Scott

MartyG-RealSense, 2020-07-21:
Thanks for the details about the interactive table.  As you are using 30' of cable, I imagined that the cameras were mounted far higher in the air than that!  The concept reminds me of an interactive RealSense table game called **Tanked!** where the RealSense camera was mounted to the side of the table's center point, atop a TV screen.

https://youtube.com/watch?v=tz0Aa4qjxeQ

I can imagine why you would want 2 cameras to cover a 12' table length.

If the 30' repeater cable length is made up of cables linked together, is there any chance of eliminating at least one of the repeater cables in the 30' chain (perhaps by moving the table closer to the NUC, or duct taping the NUC to the underside of the interactive table!)

digitaldawg, 2020-07-21:
The PC is under one end of the table.
The furthest camera is probably 24’ from the PC.
We are using only one USB extension, then the factory USB cable to the cameras.

If we could get away from this public dialogue I could send you a lot more details.

- Scott

MartyG-RealSense, 2020-07-21:
I will seek advice about whether I can conduct conversation with you through my Intel email address.

digitaldawg, 2020-07-21:
Marty -

Yesterday I received the following errors in the Viewer pop-ups:

invalid value in rs2_create_device(info_list:0F7937E8, index:0): set_xu(ctrl=1) failed! Last Error: no error

Backend in rs2_get_option(options:12C03AE0, option:Backlight Compensation hr returned: HResult 0x80004002: “No such interface supported”

Backend in rs2_create_device(info_list:0743CD18, index:1): hr returned: HResult 0x800703e3: “The I/O operation has been aborted because . . . . .

Invalid Value in rs2_create_device(info_list:0D54CE48, index:1): set_xu(ctrl=1) failed! Last Error: No Error

Scott

MartyG-RealSense, 2020-07-21:
If they are errors that only happen occasionally, I would say that they may be disconnections related to power issues during detection of the camera.  That would be consistent with the problems that you have been having with your cabling.

MartyG-RealSense, 2020-07-21:
Until you have an Intel NDA agreement, support must be conducted publicly.  I have a good impression of your table's setup based on your description though, so thanks very much for the information above.

MartyG-RealSense, 2020-08-08:
Do you have an update about your situation that you can provide, please?   Thanks!