realsenseai/librealsense #6806: rust sdk

blakat360, 2020-07-11:
Is there any chance of intel releasing a up to date wrapper library in rust?

MartyG-RealSense, 2020-07-11:
There is an unofficial librealsense2 binding for rust (**realsense-rust**) created by @jerry73204 at the link below that was updated as recently as March 2020.

https://lib.rs/crates/realsense-rust

Intel can maintain wrappers officially as part of the SDK if they are contributed as a **pull request** by the project's developer.  It is completely the developer's choice whether they do so though.

https://github.com/IntelRealSense/librealsense/pulls

blakat360, 2020-07-11:
Note this issue: https://github.com/jerry73204/realsense-rust/issues/12, the code doesn't compile.

MartyG-RealSense, 2020-07-11:
@jerry73204  Would you be able to advise @blakat360  about your rust wrapper please?  Thanks!

jerry73204, 2020-07-11:
@blakat360 thanks for your report. Looks like the build failed on ARM devices. For now, I have only Intel laptops for now, I'll check it as soon as I obtain a ARM device.

TheophileBlard, 2020-07-13:
There is also this project, by Aivero: https://gitlab.com/aivero/public/librealsense-rs
The rust implementation is very neat, however some functionalities of the SDK are still not implemented.

blakat360, 2020-07-13:
@TheophileBlard  sadly the project you linked lacks functionality I need. I'm happy to work on any of these but am not an experienced library writer so would require guidance and support.

closed 2020-08-08 by MartyG-RealSense