realsenseai/librealsense #6807: How do the depth image and color image of L515 align when rotating

wangyuanbiubiubiu, 2020-07-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |Ubuntu16.04| 
| Kernel Version (Linux Only)    | 4.15.0-107-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |v2.35.2                         | 
| Language                            |  C++                         | 
| Segment			| AR                         | 

### Issue Description
I want to use L515 for real-time 3D reconstruction, but I found that the depth image and color image could not be aligned when rotating. Is there any good solution?
It can be aligned when it's at rest：
[image]
[image]
But when you rotate it, it doesn't align：
[image]
[image]

dnjfk2003, 2020-07-12:
hi @wangyuanbiubiubiu I'm using L515 as well for get distance between camera and some object in my interest. So alignment is very important for me. But When I'm using L515 camera I knew that it doesn't align very well at the boundary pixel.(There are 10~20 pixel difference between color and depth)
I guess you should check it as well. I wrote an issue before, you can check and if you have any idea please let me know
https://github.com/IntelRealSense/librealsense/issues/6604

wangyuanbiubiubiu, 2020-07-12:
> hi @wangyuanbiubiubiu I'm using L515 as well for get distance between camera and some object in my interest. So alignment is very important for me. But When I'm using L515 camera I knew that it [...]

Thank you. I will try your question. I have also used D435i and I find that the L515 is not quite the same as D435i. When the L515 is rotated, the image of the RGB camera lags significantly behind the depth map, even with auto exposure turned off.

RealSenseSupport, 2020-07-24:
Hi @wangyuanbiubiubiu 

Have you been able to get better alignment through your efforts?

Here is a document that also may provide some information about alignment:  https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-08-13 by RealSenseSupport