realsenseai/librealsense #6810: librealsense SDK cannot open a device

abylikhsanov, 2020-07-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (from 2.35.2) | 
| Operating System & Version |   {Debian Buster (aarch64)}  | 
| Kernel Version (Linux Only)    |  (4.19)                                         | 
| Platform                                 | Custom ARM64 |
| SDK Version                            |  { legacy / 2.35.2 }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

I have built the librealsense using the cmake option `-DFORCE_RSUSB_BACKEND=ON` but when I try to run `rs-enumerate-devices` I get 
```
12/07 09:17:39,809 ERROR [281473339502136] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 12/07 09:17:39,849 ERROR [281473339502136] (types.h:307) Unable to open device interface
```
What is also interesting, when I plug my camera I get this mesg:

`[ 4817.989613] usb 1-1.2: new high-speed USB device number 9 using xhci-hcd`

Which seems to point that this is a USB2.0. But after I run my docker image with the librealsense SDK installed and try to run the executable above, I get this message:
```
[ 4897.383618] usb 1-1.2: USB disconnect, device number 9
[ 4897.790593] usb 2-1.2: new SuperSpeed Gen 1 USB device number 6 using xhci-hcd
```

What could be wrong?

tpwrules, 2020-07-17:
Just out of curiosity because I'm having a similar issue, what happens if you try `rs-enumerate-devices -d`? On my setup, regular `rs-enumerate-devices` works but adding the `-d` flag gives `RS2_USB_STATUS_BUSY` errors.

abylikhsanov, 2020-07-17:
@tpwrules This is what I get (again, at first, it said no device found, I exited the container and rerun the command):

```
Device info:
    Name                          :     Intel RealSense T265
    Serial Number                 :     952322110202
    Firmware Version              :     0.2.0.951
    Physical Port                 :     2-1.2-3
    Product Id                    :     0B37
    Usb Type Descriptor           :     3.1
    Product Line                  :     T200

Default streams:
 17/07 21:11:18,229 ERROR [281473818969152] (handle-libusb.h:95) failed to claim                                                                         usb interface: 0, error: RS2_USB_STATUS_BUSY
 17/07 21:11:18,229 ERROR [281473818969152] (types.h:307) Unable to open device                                                                         interface
2020-07-17 21:11:18,235 WARNING [default] Could not open device Unable to open d                                                                        evice interface
 17/07 21:11:18,242 ERROR [281473818969152] (handle-libusb.h:95) failed to claim                                                                         usb interface: 0, error: RS2_USB_STATUS_BUSY
 17/07 21:11:18,242 ERROR [281473818969152] (types.h:307) Unable to open device                                                                         interface
2020-07-17 21:11:18,243 WARNING [default] Could not open device Unable to open d                                                                        evice interface
 17/07 21:11:18,250 ERROR [281473818969152] (handle-libusb.h:95) failed to claim                                                                         usb interface: 0, error: RS2_USB_STATUS_BUSY
 17/07 21:11:18,250 ERROR [281473818969152] (types.h:307) Unable to open device                                                                         interface
2020-07-17 21:11:18,250 WARNING [default] Could not open device Unable to open d                                                                        evice interface
RealSense error calling rs2_config_resolve(config:0xaaab0db176b0, pipe:0xaaab0db                                                                        13f80):
    No device connected
```

Are you using Docker and ARM?

tpwrules, 2020-07-17:
No, I'm using Nix and amd64.

The disconnect and reconnect is expected. The realsense library has to upload the firmware into the T265 and reboot it each time it's plugged in. It should work the first time though. Have you tried e.g. the rs-pose sample?

abylikhsanov, 2020-07-17:
I am cross-compiling everything and simply passing the libs and binaries to the Docker container and I do not have the screen on ARM. Looks like I would need to install the shared objects on the device itself not only Docker. Sad.

tpwrules, 2020-07-17:
rs-pose is command line, it just prints the current position out in the console. The firmware is compiled into the librealsense libraries so you don't have to worry about it, and it evidently works just fine because rs-enumerate-devices couldn't have seen it otherwise.

abylikhsanov, 2020-07-17:
I see, yes it runs (displays position) but how can it help in this case?

tpwrules, 2020-07-17:
Well, what is the problem? It sounds like the camera works fine to me.

If the disconnection and reconnection is a problem, I imagine it's some quirk of your Docker setup. Bear in mind that it only has to happen the first time you access the camera after it is plugged in.

abylikhsanov, 2020-07-17:
The problem is that because my system Linux does not have any libraries for the Realsense but the Docker image has, at first, the system does not recognize the camera and does not load any libraries so therefore, when I start the Docker container it also does not see the camera. Then, as you already know, I run a sample realsense binary which loads the hardware and detects it but sadly, I then have to exit the container and reenter, after which the container starts to see the camera straight away.

tpwrules, 2020-07-17:
This would happen even if Docker were not involved and can't be mitigated because the T265 has no firmware storage. It has to be loaded by the host.

The problem is apparently that Docker doesn't rescan for USB devices unless you restart it. The T265's VID:PID after the firmware is loaded is 8087:0b37 (and 03e7:2150 before). Maybe you can add a rule to allow those all the time? I don't know anything about how Docker works.

closed 2021-03-18 by RealSenseSupport