realsenseai/librealsense #6811: using a.bag frames

arothenberg, 2020-07-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
I am trying to analyze depth frames on  prerecorded data.

To test it I. using the rs-align-advanced code and changing the source to the "a.bag" file created using rs-record-playback .
I tried to port this code
  cfg.enable_device_from_file("a.bag");
 into rs-align-advanced like this

    rs2::pipeline pipe;
 rs2::config cfg;
  cfg.enable_device_from_file("a.bag");
  rs2::pipeline_profile profile = pipe.start(cfg);

I get an error saying failed to create a ros reader. 

The difference i see is that  rs-record-playback uses a shared pointer 
  pipe = std::make_shared<rs2::pipeline>();
...
   pipe->start(cfg); 

I'm probably missing something obvious. 

Thanks in advance. Also, is the bag file degraded or does it store the same frames that the camera sends?

kograins, 2020-07-13:
To arothenberg:

I am using python API, The method "config.enable_stream_from_file("filename")" didn't worked well  with the loaded .bag filename  named with prefix like "test....", "2020....".

MartyG-RealSense, 2020-07-13:
Hi @arothenberg   If you are working in the C++ language, the **rosbag reader** sample program created by a RealSense community member may be a useful reference.  It also requires OpenCV as a dependency though.

The finished code for the project is at the bottom of the opening comment, in a file called **RealSenseHelper.zip**.

https://github.com/IntelRealSense/librealsense/issues/2215

arothenberg, 2020-07-15:
Thanks Marty. 
I downloaded the zip and will see what it's all about. I didn't want to add the dependency but I guess Opencv is necessary for certain tasks.  

Thanks. I'll open another issue or comment on the marcovs thread if(when!) I have a problem. Thanks.

closed 2020-07-15 by arothenberg

MartyG-RealSense, 2020-07-15:
@arothenberg   Thanks very much for the update.   Good luck!

arothenberg, 2020-07-15:
Thanks. One other thing. Realsense should consider adding the parser or similar to the sdk examples because testing with  prerecorded data is necessary for most projects.

MartyG-RealSense, 2020-07-15:
The rosbag parser program is at the base of the SDK examples page, filed under the **Community Projects** section of example programs contributed by RealSense community members.

https://github.com/IntelRealSense/librealsense/tree/master/examples#community-projects

arothenberg, 2020-07-15:
I got it thanks. I'll set it up today. I never thought of checking the community projects. I'll be more aware of that now. It probably makes things easier.

arothenberg, 2020-07-17:
Hey Marty,

I'm  downloading  ROS and  it's dependencies from this site
http://wiki.ros.org/Installation/Windows

It's a long download. It installs a dozen distributions. It isn't huge- about 1/2 gig I think - but it's very involved. lots of stuff being installed on my pc. 
I'll hack through it and hopefully be able to use the marcov solution. I noticed some on the morcov thread were having problems setting up rosbag in their environment. I'll see what happens.

Is the marcov solution currently the best way to consume bag files? Is there a simpler method that uses another format (not bag) ? I have no problem trying to make this work but I want to make sure this is the right path because it could take days?

Thanks.