realsenseai/librealsense #6812: Get guaranteed time interval while getting frames and image processing with MATLAB.

mk-kor, 2020-07-13:
----------------------------------------------------------------------------------------------------

| Required Info                         | Get guaranteed time interval while getting frames and image processing |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |    | 
| Platform                                 | PC |
| SDK Version                            |  2.35.2  | 
| Language                            |  Matlab                           | 
| Segment			| Robot                     | 

### Issue Description
Hi.

I'm currently developing vision system for robot using Realsense D415.

I need to get time interval which should be constant for my mathematical model.

And I'm sure 'wait_for_frames()' function gave me frames every constant time according to FPS. (Assume that there is no frame drop)

But while image processing, time became variable.

I tried to resolve the problem using parallel computing toolbox(parfor, spmd, use GpuArray and so on), but it doesnt work.

I know it is inevitable, and my question sounds weird.....

Could u give me any advice?? Should i use c++ for multi thread processing??

Thanks in advance!! :)

RealSenseSupport, 2020-07-28:
I think you can use the metadata timestamp instead of the time of wait_for_frame.
you can try `depth.get_frame_metadata(realsense.frame_metadata_value.time_of_arrival);` for query the timestamp.
https://github.com/IntelRealSense/librealsense/blob/v2.37.0/wrappers/matlab/frame.m#L35
https://github.com/IntelRealSense/librealsense/blob/v2.37.0/wrappers/matlab/frame_metadata_value.m#L10

And don't forget enable the metadata in the OS.
https://github.com/IntelRealSense/librealsense/blob/v2.37.0/doc/installation_windows.md#update-registry

RealSenseSupport, 2020-09-04:
@mk-kor Any other questions about this? Looking forward to your update. Thanks!

RealSenseSupport, 2020-09-24:
@mk-kor If no more questions about this, will close it accordingly. Thanks!

closed 2020-09-24 by RealSenseSupport