realsenseai/librealsense #6813: [Question] Does not support multicamera for Unity?

wodndb, 2020-07-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.35.2 | 
| Language                            | unity | 
| Segment			|  others | 

I develop multicamera application using unity.

However I see that realsense SDK does not support multicamera in support feature matrix. ([link](https://dev.intelrealsense.com/docs/supported-features-matrix)).

But multicamera works well in unity.

So what is mean that multicamera support?

MartyG-RealSense, 2020-07-13:
Hi @wodndb   RealSense SDK 2.0 has had multicam support since it was first launched in late 2017 (hardware sync between multiple cameras came a bit later after launch).

In regard to multicam in Unity, I'm not aware of officially announced support for it, though it has certainly been done in the past by RealSense community members.  In June 2020, RealSense community member **JMcC16** kindly shared on the Intel Support RealSrnse forum their method for using multiple cameras in the RealSense Unity wrapper.

https://support.intelrealsense.com/hc/en-us/community/posts/360047049173

wodndb, 2020-07-13:
@MartyG-RealSense Thanks. I understand that RealSense SDK unofficially support multicamera streaming in unity.

closed 2020-07-13 by wodndb