realsenseai/librealsense #6814: L515 "Receiver Gain" slider backwards

drewm1980, 2020-07-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.36.0 }                          | 
| Language                            |  { Rust }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

I am trying to image objects that have high IR reflectivity at very close range. I am seeing an artifact that is clearly caused by specularities caused by retro-reflection of the IR laser. I confirmed "0xff" pixel values in the infrared stream where the artifacts occur. The observed effect is an under-estimation of depth at the specularity, causing "spikes" towards the L515 in the 3D data.

I confirmed the saturation in the IR image.
I started with the "Short Range" preset.
I changed the "Laser Power" setting to "0.0" (which is apparently not  actually zero, so that bound should be adjusted too).
I changed the "Min Distance" setting to 0
I am experimenting with the "Receiver Gain" setting and getting surprising behavior.

My expected behavior for the "Receiver Gain" slider is the IR stream should increase as the "Receiver Gain" setting slider increases, but I am seeing the opposite. However, in order to achieve the lowest gain (in this case to combat the specularities), I must use the highest Receiver Gain setting.

Thanks!

dorodnic, 2020-07-13:
You are correct in your observation. 
Personally, I'm not sure if its a good idea to invert it at this point, since this may break functionality

drewm1980, 2020-07-13:
Would calling it "(Inverse) Gain" be a non-breaking change?

This device only just became available here in Belgium; you can go ahead and make breaking changes up until ~my project is in production ;)

RealSenseSupport, 2020-12-08:
I wanted to come back to this request.  We've done some modifications to our Viewer in the latest release - https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0

We modified the "Digital Gain" option.  I've included the release notes here:  https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

There are a lot of updates for L515 in this latest release.

We reviewed the "Receiver Gain" slider and will keep it as it is.  As dorodnic stated at this time we shouldn't modify it as it could cause other issues.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-21 by RealSenseSupport