realsenseai/librealsense #6816: L515 extended settings to combat IR saturation

drewm1980, 2020-07-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.36.0                          | 
| Language                            |                       | 
| Segment			|       Industrial Robotics                  | 

### Issue Description

I am trying to image spherical objects that have high IR reflectivity (a kind of fruit), at close range. The returned IR image has saturated values at the pixels corresponding to direct retroreflection from the laser the photodiode. I have confirmed the "0xFF" pixel values in RealSenseViewer. This is causing underestimates of the depth at the saturated pixels, that appear as a "spike" towards the camera.

It's not important for my application, but I observe a similar behavior behavior when pointing the camera at a laptop screen. In this case at the specularities the depth is ~overestimated, causing a spike away from the camera.

To better handle this situation, would any of the following be possible?
* Extend the "Laser Power" setting on the low end. I currently have it set to "0", but that does not actually turn off the laser.
* Allow the "Receiver Gain" to specify a lower gain (i.e. reduced photodiode sensitivity). (See related issue https://github.com/IntelRealSense/librealsense/issues/6814 regarding the numerical values being seemingly backwards)
* If this can't be solved in software/firmware, should I try a neutral density filter or gel over the IR receiver? 

* As a mitigation, stop "hole filling" the values where the IR stream saturated. I can do that in my own post-processing, but it would be more elegant for the camera to just not return values at the saturated pixels. Maybe this is already do-able with the "Confidence threshold" setting, but I am unable to set it high enough; perhaps the threshold could be changed to go higher than 3?

RealSenseSupport, 2020-07-23:
@drewm1980 

We are looking into this.

closed 2020-07-23 by RealSenseSupport

Smith-Lachlan, 2021-03-17:
Any progress on this?