realsenseai/librealsense #6817: Delay for camera preview in New android wrapper.

ankittecholution, 2020-07-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |  Android  | 
| Platform                                 | Android |
| SDK Version                            |  2.31.0, 2.35, 2.32.5 , 2.36.0                           | 
| Language                            |  Android java wrraper                         | 

### Issue Description
I have seen there a lag in all the new version from v2.31.0 in android wrapper i started working on intel realsense with 1280,720 color and depth frames it working fine in camera perivew without lag if i upgrade the library to 2.35 or 2.35.2 or 2.36.0 it still laggy. The preview from the camera is not smooth it has a delay.

MartyG-RealSense, 2020-07-15:
Hi @ankittecholution   Is the delay that you are experiencing occurring with the RGB, please?  There was a bug-fix for a preview for the RGB stream added in SDK **2.35.0**:

**Android: Fix YUYV preview for the RGB stream in camera app.**
https://github.com/IntelRealSense/librealsense/pull/6143

It would be useful if you could test the preview with earlier SDK versions such as **2.34.0** to see whether the lag begins at SDK 2.35.0 onwards.  Thanks!

ankittecholution, 2020-07-15:
There was lag when i start both depth and colour frame in preview.

ankittecholution, 2020-07-15:
Sure will test it and update the thread.

MartyG-RealSense, 2020-07-15:
Thanks so much, I very much appreciate the effort you are making.   :)

kafan1986, 2020-07-28:
@MartyG-RealSense There is certain regression in the camera stream. I am using RGB and Depth streams and notice delay and intermittent choppiness of RGB stream (probably due to frame skips). I am using camera at 15 FPS. The delay is around 1.5 to 2 seconds.

I downgraded my SDK to v2.30.0 and firmware to 5.12.1.0 but the issue remained.

MartyG-RealSense, 2020-07-28:
@ankittecholution  @kafan1986   As more than one person has reported the same issue with lag when using depth and RGB together on the Android wrapper, I will file an official bug report with Intel now so that it can be investigated.  This case should be kept open whilst the report is being investigated.

kafan1986, 2020-07-28:
I am not sure whether the source of problem is the firmware related or SDK related and also I am not sure whether I am noticing the delay now more or whether it always existed.

@ankittecholution Did you find any combination of SDK/firmware that resolved the issue for you?

ankittecholution, 2020-07-28:
Yes as I mentioned it always come with any SDK 2.35 or 2.35.2 or 2.36.0 but not in 2.31

kafan1986, 2020-07-28:
> Yes as I mentioned it always come with any SDK 2.35 or 2.35.2 or 2.36.0 but not in 2.31

What firmware version are you using?

ankittecholution, 2020-07-29:
Are you asking about the camera Firmware???

kafan1986, 2020-07-29:
> Are you asking about the camera Firmware???

Yes

ankittecholution, 2020-07-29:
Let me check right away

ankittecholution, 2020-07-29:
[image]
I think this wat you were asking right?

kafan1986, 2020-07-29:
Yes. The firmware you used is quite recent one 5.12.02.100. Although the latest firmware is 5.12.6

joeyf123, 2020-07-29:
I was having the same issue as you and on updating to the latest my problem went away.

kafan1986, 2020-07-29:
I still had lag. Although somewhat minor. I lowered the RGB resolution to 480p and helped reduce the lag.

kafan1986, 2020-07-29:
btw, has anyone of you tested the depth camera as just a simple camera on android? Does it work like that?

joeyf123, 2020-07-29:
Yes, The realsense makes a fairly good camera when you are using the RGB camera and the Color StreamType.

kafan1986, 2020-07-29:
I am asking whether the camera gets detected as a regular camera (webcam) for android native camera 1 or 2 API or does it always need the SDK and specific code to get the frame data? I believe the D4xx series camera gets detected as regular RGB camera on Linux and maybe on windows platform.

ankittecholution, 2020-07-29:
No, I didn't try only the depth camera.

ankittecholution, 2020-07-29:
So I think you guys are saying if I update the firmware it would resolve the issue???

MartyG-RealSense, 2020-08-15:
@ankittecholution    Do you still require assistance with this case, please?   Thanks!

ankittecholution, 2020-08-15:
Is the delay resolved in new release.

MartyG-RealSense, 2020-08-15:
Never mind, I just read back through the case and reminded myself that the case needs to be kept open because there is still an active bug report related to it that has not been resolved yet.  My apologies.

No, I have no updates to provide.  Thanks for your quick response.

ankittecholution, 2020-08-29:
@kafan1986 I think this delay will not happen in Release mode it will only happen in the case of debug mode. I have not tried it but if anyone else can try and give a report it will give a clear idea for the issue. If the Frame doesn't lag in release mode then I think we can close this issue.

MartyG-RealSense, 2021-02-13:
Cased closed due to no further reports of the problem received for months and the likelihood that it occurs only in Debug mode.

closed 2021-02-13 by MartyG-RealSense