realsenseai/librealsense #6820: AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

skywo1f, 2020-07-13:
Pyrealsense2 doesnt seem to be able to find pipeline. I am running this on a rpi 4 with 64bit os. I originally tried to run it on the regular 32 bit rasbpian, but apparently the L515 is not compatible with 32 bit operating systems.  This error occurs when trying to run the opencv_viewer_example.py script in ~/librealsense/wrappers/python/examples . Had no issues during installation and I compiled librealsense with the python flag. (-DBUILD_PYTHON_BINDINGS:bool=true). Should I have compiled librealsense with any other flags?

prodrigovalero, 2020-07-13:
I have the same issue with jetson nano.

RealSenseSupport, 2020-07-16:
Please try to copy the new library to the same folder of your script to see if help
it could be the path of the library generated with SDK2.36 is changed.

wwppyy

RealSenseSupport, 2020-07-16:
you can refer to 
https://github.com/IntelRealSense/librealsense/pull/6220

mkmeral, 2020-07-17:
I have the same problem and it does not help, I don't think it's generating bindings for python3, even when i tried to force with -DPYTHON_EXECUTABLE. Here are all the files under /usr that contain pyrealsense in their name (used `find . -wholename '*pyrealsense*'` as a command). 
`./usr/local/lib/cmake/pyrealsense2
./usr/local/lib/cmake/pyrealsense2/pyrealsense2Targets-noconfig.cmake
./usr/local/lib/cmake/pyrealsense2/pyrealsense2Targets.cmake
./usr/local/lib/cmake/pyrealsense2/pyrealsense2ConfigVersion.cmake
./usr/local/lib/cmake/pyrealsense2/pyrealsense2Targets-release.cmake
./usr/local/lib/cmake/pyrealsense2/pyrealsense2Config.cmake
./usr/local/lib/python2.7/pyrealsense2
./usr/local/lib/python2.7/pyrealsense2/pyrealsense2.so.2.36.0
./usr/local/lib/python2.7/pyrealsense2/pyrealsense2.so.2.36
./usr/local/lib/python2.7/pyrealsense2/pybackend2.so
./usr/local/lib/python2.7/pyrealsense2/pyrealsense2.so
./usr/local/lib/python2.7/pyrealsense2/pybackend2.so.2
./usr/local/lib/python2.7/pyrealsense2/pybackend2.so.2.36.0`
I tried exporting  /usr/local/lib/cmake/  to pythonpath, but got no attributte pipeline error. 

Also I have tried cd into librealsense/build/wrappers/python/ (the build folder which contains pyrealsense2.so, pyrealsense2.so.2.36, pyrealsense2.so.2.36.0) and use python3 and import pyrealsense2, then I got 
`ImportError: /home/username/librealsense/build/wrappers/python/pyrealsense2.so: undefined symbol: _Py_ZeroStruct`

tpwrules, 2020-07-17:
I found an issue with the installer failing to copy the correct __init__.py into the installed python module. This meant you just imported an empty directory.

Try copying [this](https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/wrappers/python/pyrealsense2/__init__.py#L1) file in the repo to the directory with the installed .so files, in your case `/usr/local/lib/python2.7/pyrealsense2`. If you're getting an ImportError like that though, something may have gone wrong during building and this solution probably won't help.

jpdevicente, 2020-07-24:
Im having this exact same issue on a jetson xavier. I tried coping the __init__.py, librealsense2.so and pyrealsense2.cpython-36m-aarch64-linux-gnu.so to the same folder where my scripts are, but I get the error: ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2

When I do the same without coping the __init__py I get AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

Any known solution to this?

bobcassels, 2020-07-27:
When I built v2.36.0 for Jetson Nano like this:
```
/usr/bin/cmake ../ -DBUILD_EXAMPLES=true -DFORCE_LIBUVC=true -DBUILD_WITH_CUDA=true -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=bool:true
```
I got no `pyrealsense2` module at all. I tried again, adding `-DPYTHON_EXECUTABLE=/usr/bin/python3`. That built an empty pyrealsense2 module, getting the error in the subject of this issue.

It would be nice to have a fix for this, since v2.34.0 has roughly 1-second lag in the camera image. Right now, the camera is pretty much useless. When I built v2.34.0, I did not need `-DPYTHON_EXECUTABLE=/usr/bin/python3`. It would be nice to know how I'm supposed to build, to get a Python 3 `pyrealsense2` module.

skywo1f, 2020-07-27:
also the raspberry pi does not detect the l515 when I run realsense-viewer

skywo1f, 2020-07-27:
> Please try to copy the new library to the same folder of your script to see if help
> it could be the path of the library generated with SDK2.36 is changed.
> 
> wwppyy

I tried moving the opencv examples back a folder to the rest of the pyrealsense2 library, but it is now telling me "No module named 'pyrealsense2.pyrealsense2'

skywo1f, 2020-07-27:
> you can refer to
> #6220

I dont understand what this has to do with our error

bobcassels, 2020-07-30:
A friend of mine figured out a hack to get `pyrealsense2` to build. He got it to work in v2.36.0. I successfully built v2.37.0. After you check out the sources you want to build, replace one file with an older copy:
```
git checkout v2.34.0 -- wrappers/python/CMakeLists.txt
```

jpdevicente, 2020-07-30:
That works, your friend is a hero. Thank you

skywo1f, 2020-07-31:
using the CMakeLists.txt trick on the rpi4 did not work for me. However, using cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true  
at least lets me see the device in the realsense viewer

skywo1f, 2020-07-31:
all I see in rs is :
>>> vars(rs)
{'__name__': 'pyrealsense2', '__doc__': None, '__package__': 'pyrealsense2', '__loader__': <_frozen_importlib_external._NamespaceLoader object at 0x7f9f8b7550>, '__spec__': ModuleSpec(name='pyrealsense2', loader=<_frozen_importlib_external._NamespaceLoader object at 0x7f9f8b7550>, submodule_search_locations=_NamespacePath(['/usr/lib/python3/dist-packages/pyrealsense2'])), '__file__': None, '__path__': _NamespacePath(['/usr/lib/python3/dist-packages/pyrealsense2'])}

bobcassels, 2020-08-01:
@skywo1f Your result is consistent with using a post-v2.34.0 version of the `wrappers/python/CMakeLists.txt` file. I suggest you verify that you really have the old version of that file, during the build. It took me a couple of tries. The build script I was using checks out a version of the source in two different places. I had to add my checkout of the older version of that one file, in both places. And then I verified during the build that the old version of the file was still present.

skywo1f, 2020-08-03:
So if I move my scripts into the /librealsense/build/wrappers/python folder, and re-install my opencv then it works. I also had to change the resolution in the example to 1024 by 768

closed 2020-08-03 by skywo1f

JerryLee333, 2020-09-24:
@skywo1f  Hello , I'm facing the smae problem current and don't know how to fix it .

AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

I try to put the script in the  /home/pi/librealsense/build/wrappers/python/   , like what you do in the end

and run it , it seems it can find pipeline , but another problem show up 

it can't find CV2  

pi@raspberrypi:~ $ python3 /home/pi/librealsense/build/wrappers/python/align-depth2color.py
Depth Scale is:  0.0010000000474974513
Traceback (most recent call last):
  File "/home/pi/librealsense/build/wrappers/python/align-depth2color.py", line 73, in <module>
    cv2.imshow('Align Example', images)
cv2.error: OpenCV(4.4.0) /tmp/pip-install-gunf1z13/opencv-python/opencv/modules/highgui/src/window.cpp:651: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvShowImage'

Thank you to sharing the info  .

JerryLee333, 2020-09-24:
@skywo1f  

pi@raspberrypi:~ $ python3 /home/pi/librealsense/build/wrappers/python/align-depth2color.py
Depth Scale is: 0.0010000000474974513
Traceback (most recent call last):
File "/home/pi/librealsense/build/wrappers/python/align-depth2color.py", line 73, in
cv2.imshow('Align Example', images)
cv2.error: OpenCV(4.4.0) /tmp/pip-install-gunf1z13/opencv-python/opencv/modules/highgui/src/window.cpp:651: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvShowImage'

After this problem , I try to reinstall the opencv , and things got solved . 

Thank you for sharing the repo .

mirkocomparetti-synesis, 2021-04-19:
Hello @RealSenseSupport 
I see this one is closed, but we are facing the same issue on the rpi4 with the 2.42 version. Are you guys aware of something like this?

Best

Somroy1993, 2021-09-14:
I tried with latest build following official docs. Ignore protobuf, cmake and opencv instalation guide from the doc. The problem was if you pip install in ubuntu you have to import it like import **pyrealnsense2 as rs** .  But when u build it from source you have to import it like **import pyrealsense2.pyrealsense2 as rs** . 
Try like this.
>>> import pyrealsense2.pyrealsense2 as rs
>>> rs.__version__
'2.49.0'
>>> 
Let me know if u have any issue regarding dependencies.

yosis1969, 2022-08-12:
> I tried with latest build following official docs. Ignore protobuf, cmake and opencv instalation guide from the doc. The problem was if you pip install in ubuntu you have to import it like import [...]

Yup did work! 
import pyrealsense2.pyrealsense2 as rs