realsenseai/librealsense #6822: Are the mount and calibration matrix in the tracking and depth example compatible with the d415 ?

Erandurthil, 2020-07-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 & T265 } | 

I'm trying to find out if the provided .stl and the transformation matrix corresponding to it from the example tracking and depth is compatible with the D415. I already printed and mounted the cameras, I'm just not sure if the Matrix is valid for all D4xx cameras. 
If not, is there one out there ? Or would I need to measure/calculate that myself  to be able to transform the point data to "world" coordinates.

dorodnic, 2020-07-15:
I think the mount was only designed for the D435 so it can be used as a starting point but likely to require some modifications

Erandurthil, 2020-07-15:
I have already looked into the datasheets and physcial dimensions of all three cameras and there a definitely some physical differences between the D435 and D415, which would at least mandate a change in the translation of the transformation matrix. Since the handedness of the coordinate systems of the depth cameras frames are identical, I'd assume the rotational part is still valid. 

Sadly the technical drawing in the T265 datasheet isn't really readable, is there a higher resolution one availble anywhere ?

[image]

Also the datasheet of the D415 only mentions it's X and Z coordinates/measurements in regards to the frame, I am correct to assume that the sensor frames origins Y coordinate is the exact middle of the housing of the camera ? 
[image]
[image]

RealSenseCustomerSupport, 2020-07-16:
----------------------------------------------

Hi Erandurthil,
We have stp 3D file published on
https://dev.intelrealsense.com/docs/cad-files
you can measure the mentioned you are interested.

Erandurthil, 2020-07-16:
Those are great, thanks. 

Last piece I am missing is the Z measurement for the tracking camera frame. Does it start at the glass cover of the housing or at the module itself ? 
It seems in like it isn't explizitly stated which assumption is correct, but since the distance from the housing to the module doesn't seem mentioned I am assuming it's the first one ?

Erandurthil, 2020-07-17:
[T265_D415.zip](https://github.com/IntelRealSense/librealsense/files/4938296/T265_D415.zip)

Based on the datasheets and the cad models I designed a new mount and a transformation matrix, which seem to be working good enough.

closed 2020-07-17 by Erandurthil