realsenseai/librealsense #6823: What is the Preset Enum for L515?

FLS-BP-US, 2020-07-13:
What is the preset Enum for the  L515?

RealSenseSupport, 2020-07-14:
Hi @FLS-BP-US 

Can you provide an image that shows this preset?  Is this seen with the latest SDK 2.36.0 and latest FW that is available?

FLS-BP-US, 2020-07-14:
I have install the latest sdk and firmware from the website. It say it is 2.35.2.1897.
[image]

RealSenseSupport, 2020-07-14:
Hi @FLS-BP-US 

Do let us know what you see after building and launching with the main branch 2.36.0 release.  

In the image I don't see the "Preset Enum" that you oultine.  I also do not see it on my end either.  Just want to have an idea of what you're seeing so that we can help you better.

FLS-BP-US, 2020-07-14:
[image]

dorodnic, 2020-07-15:
Hi @FLS-BP-US 
Enum - [rs_option.h:138-148](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_option.h#L138-L148)
At the moment, depth units are not configurable. We may introduce the option to change them in future releases, similar to how it was done for the D400.

FLS-BP-US, 2020-07-15:
Hi @dorodnic,
Thank you, that solves the question.  Do the depth units impact the camera intrinsics? If I want to scale depth image to have units of millimeters, can I just divided the image matrix by 4 or am I oversimplifying things? I want to comparing the 435i and L515 as a wear measurement device.

closed 2020-07-17 by FLS-BP-US

dorodnic, 2020-07-22:
Hi @FLS-BP-US 
Sorry I missed your follow-up question. Depth units are orthogonal to camera intrinsics as far as I know. I think scaling by 4 is Ok, but I'd recommend using depth units from the SDK or `rs2::units_transform` class. This future proofs the code to work with any other RealSense depth camera. Also, while depth units of the L515 are fixed at the moment, we did receive user request to make them configurable and will get to it once more severe issues are resolved.