realsenseai/librealsense #6824: DOCUMENTATION: Capturing SLAM_ERROR in T265

marcelino-pensa, 2020-07-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-106-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.34.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot/Drone                          | 

### Issue Description

I've been using a T265 with `realsense-ros` in `2.2.13`, and I've noticed that it will sometimes lose localization on my drone. I have failsafes to capture if `/camera/odom/sample` stops publishing, but it can take many seconds after a SLAM_ERROR occurs before it stops publishing, which can be catastrophic for the drone flight itself. 

I would like to capture SLAM_ERROR instances, which are not captured in `realsense-ros`. I went through [this link](https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme), but it seems insufficient for me to get started. It might sound trivial, but the documentation above does not give us any clue on how to declare the variable `dev`. Which datatype is `dev`? Also, which libraries need to be included in the file that I declare the `dev` variable? 
Also, will the examples in [the link above](https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme) even capture SLAM_ERROR? Based on the types of errors, it doesn't even seem like SLAM_ERROR would be captured...

Note: I apologize for creating a ["duplicated issue"](https://github.com/IntelRealSense/realsense-ros/issues/1264), but I haven't received any answer so far in `realsense-ros`, I am hopeful that this will be answered here.

dorodnic, 2020-07-15:
Hi @marcelino-pensa 
[ar-advanced](https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-advanced) shows how you can register to hardware events from the T265, but I'm not sure SLAM_ERROR is raised via this API (its mainly used for relocalization events)
Another way is to catch all SDK warnings - 
```
rs2::log_to_callback( RS2_LOG_SEVERITY_WARNING,
        [notifications]( rs2_log_severity severity, rs2::log_message const& msg )
        {
            if(msg.raw().find("SLAM_ERROR") != std::string::npos)
            {

            }
        });
```

marcelino-pensa, 2020-09-01:
Hi @dorodnic! I am looking into this issue, and I was wondering if this would be interesting to become a PR that would merge back to `realsense-ros`.
If you believe so, then I'd have the following questions:
1) Whenever I capture an error, is it okay to publish it into a topic?
2) If so, what topic would it be? I noticed that realsense2_camera publishes into the topic `diagnostics`, which seems reasonable for the purpose of this issue.
3) If you think that `diagnostics` is not a reasonable topic for this purpose, which topic (and what data type) would be reasonable for publishing this?

dorodnic, 2020-09-08:
Hi @marcelino-pensa 
Pull-request would be welcomed. I think this approach is good, however, I do not have enough experience with ROS, @doronhi is the maintainer, I'll check with him

marcelino-pensa, 2020-09-08:
thanks, @dorodnic! I'll be waiting for an update in here!

marcelino-pensa, 2020-09-21:
closing due to the merge above

closed 2020-09-21 by marcelino-pensa

hfdlwt, 2023-07-28:
Hello ! @marcelino-pensa  I have reviewed the [link](https://github.com/IntelRealSense/realsense-ros/pull/1394/commits/013aed0487fb2c6705500b17e3e2b219ad28bbd9) you provided, and do I  still need to add any modifications? Ability to directly read SLAM_ERROR information through topics?