realsenseai/librealsense #6825: Depth to color frame from jpeg and png files

zc-alexfan, 2020-07-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Operating System & Version |  Linux (Ubuntu 14/16/17) / MacOS  | 

### Issue Description
Hi,
I am interested in an application that requires alignment of the color and depth images but it seems that they are shot in different coordinate frames. Unfortunately, the color images and depth images are stored in jpeg and png files respectively. I am wondering if there is a way to map depth images to the color image frame?

I tried looking into the realsense github page, but the depth2image interface for that library is only for either bag files (raw realsense data file) or streaming data. It would be nice if you could point me to either a code base or something else.

I am unfamiliar with camera transform. I don't know if it is possible to do the mapping using the measured intrinsics and extrinsic parameters?

MartyG-RealSense, 2020-07-14:
I hope that the discussion in the link below will provide useful guidance about aligning a color image file with a depth image file.

https://github.com/IntelRealSense/librealsense/issues/1274

zc-alexfan, 2020-07-14:
Thanks a lot! I coded some python code by reading at the source code of realsense. In particular, I deprojected the depth, change the coordinate frame and project to the color frame.

However, it seems that the depth frame has much smaller resolution, which creates some "holes" when it is projected to a higher resolution RGB image. Is there a standard way to fill in the holes? My current thought is to perform some form of bilinear interpolation, but the new depth image is so sparse that regular bilinear interpolation might not make sense.

A resultant image is attached: https://ibb.co/4dq14pn

MartyG-RealSense, 2020-07-15:
I considered this problem carefully.  The SR300's depth resolution is **640x480** VGA.  This suggests to me that the ideal RGB resolution would be 640x480 (also VGA).    

[image]

Would it be possible to manually resize the RGB image in image editing software to match the resolution of the depth image file, or are there too many RGB images in your set to make this practical?  If you are using Linux, you may be able to do a **batch resize** operation on the files.  Please google for **linux batch resize images** for research leads about how to do this.  Here is an example from such a search:

https://guides.wp-bullet.com/batch-resize-images-using-linux-command-line-and-imagemagick/

zc-alexfan, 2020-07-15:
Thanks a lot! That makes sense. I am going to close this issue. For anyone who is interested [this page](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0) shows the basics operations one needs to do to perform the alignment. Following up the link on that page, you can look up the C++ source code and reimplement the projection in any language you prefer.

closed 2020-07-15 by zc-alexfan