realsenseai/librealsense #6826: Is there any api to control IR LED on / off

superxi, 2020-07-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | F416 | 
| Operating System & Version |   ubuntu 16.04  | 
| Platform                                 | PC |
| Language                            |  python                         | 

### Issue Description
Is there any api to control IR LED on / off

dorodnic, 2020-07-15:
Hi @superxi 
Yes - [RS2_OPTION_EMITTER_ENABLED](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_option.h#L42)
3 to enable, 0 to disable.

RealSenseCustomerSupport, 2020-07-21:
----------------------------------------------

Hi @superxi Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2020-08-11 by dorodnic