realsenseai/librealsense #6827: L515 is worse than D345 for alignment performance

DreamLFairy, 2020-07-14:
This is the alignment effect of the d345.
The resolution of the color map is 1280*720，The resolution of the depth map is 1280*720。
 [image]

This is the alignment effect of the L515.
The resolution of the color map is 1024*768，The resolution of the depth map is 1280*720。
Resolution of the red point cloud  1280*720，and resolution of the yellow point cloud  1280*720.

[image]

The two point clouds(red and yellow) are taken by the same camera at the same time, so the alignment effect of L515 should be the same as that of the d345.

This question has been asked before, but no solution has been found. Now I want to ask again, the reason why the alignment effect of l515 is poor is the hardware (for example, the effect of internal and external calibration is not good?) Or software?

Or there are other solutions？

RealSenseSupport, 2020-07-22:
@DreamLFairy  What SDK and FW version do you use for L515? Also double check my understanding of your L515 configuration.  Is that depth 1024x768 and color 1280x720?

RealSenseSupport, 2020-08-12:
@DreamLFairy Could you please update if you still need support on this? Thanks!

RealSenseSupport, 2020-08-21:
@DreamLFairy Could you please update? Thanks!

RealSenseSupport, 2020-09-04:
@DreamLFairy  Sorry that we didn't hear from you for weeks. Will close this accordingly. Please feel free to create another ticket if you still have issues or questions. Thanks!

closed 2020-09-04 by RealSenseSupport