realsenseai/librealsense #6829: Calibration of D435 and D415 in Unity

fajarnugroho93, 2020-07-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Specifically D435 and D415 | 
| Operating System & Version |   Win 10  | 
| Platform                                 |  PC  |
| SDK Version                            |  2                         | 
| Language                            |  Unity                         | 

Hello,

I successfully setup my scene to be able to do live stream from D435 and D415 at the same time based on https://github.com/IntelRealSense/librealsense/issues/6813 which pointed me to https://support.intelrealsense.com/hc/en-us/community/posts/360047049173

But the point cloud renderings are separated from both of the cameras. Is there a way to properly combine them? Is calibration even the right term? Is it align?

The example from https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam mentioned something with `rs2::context`, I find similar thing in Unity wrapper but I don't really understand what it does. It used in `RsDeviceListener` and passed to `Pipeline`. Is it related?

I have seen a link to https://communities.intel.com/message/558223#558223 from https://github.com/IntelRealSense/librealsense/issues/2203 which seems to be a help, but it's gone now.

MartyG-RealSense, 2020-07-14:
Hi @fajarnugroho93   The broken link discussed this unofficial extension of the RealSense Unity wrapper:

https://github.com/ajingu/RealSense

It is so out of date compared to the current official wrapper now though that it may be of limited use other than a reference.

The subject of combining point clouds in Unity was discussed in March 2020 in the link below:

https://github.com/IntelRealSense/librealsense/issues/6147

A discussion in the link below outside of the RealSense forums also looks in detail at how a multicam RealSense / Unity setup might combine point clouds.

https://forum.commonvisionblox.com/t/combining-point-clouds/908/3

fajarnugroho93, 2020-07-14:
Hi @MartyG-RealSense 

Thank you for pointing me to the other issue, I asked there.

closed 2020-08-02 by MartyG-RealSense