realsenseai/librealsense #6831: apt packages for bionic no longer working

jackjansen, 2020-07-14:
Installing the apt packages (in a Docker image running ubuntu 18.04, in my case) no longer works. It worked less than a week ago.

I've re-installed the apt-keys and re-added the repo, with

```
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
RUN add-apt-repository -y "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u 
```
but it doesn't help.

The error on `apt-get update` or `apt-get install librealsense2-dkms` is:

```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.21.201 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
```

danielgordon10, 2020-07-14:
Broken for me as well.

ev-mp, 2020-07-14:
Thanks for the report,  the public server is indeed not accessible during the last  hour.
We're troubleshooting the issue and we'll update accordingly.

ev-mp, 2020-07-14:
All server operations were resumed.
 @jackjansen ,@danielgordon10 - pls check now.

danielgordon10, 2020-07-14:
Confirmed working again. Thanks.

ev-mp, 2020-07-14:
Thanks again. closed as resolved.

closed 2020-07-14 by ev-mp