realsenseai/librealsense #6833: Setting controls/options in Python

jpenaloza1211, 2020-07-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435e } | 
| Firmware Version                   | (05.12.03.00) | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1                         | 
| Language                            |  {python 3.7}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
These might be some simple questions, but I just want to make sure that I can optimally tune settings and haven't found exact answers anywhere yet. 
 Question 1: If I use the default streaming method as is the case with the frame alignment example provided in this repository (align-depth2color.py), will the camera use automatic settings (like exposure time, etc)?
 Question 2: I know that in the viewer, I can play around with those settings, but is there any tutorial on adjusting those settings in python?
Thank you so much!

MartyG-RealSense, 2020-07-15:
Hi @jpenaloza1211   If you are using the **FRAMOS D435e**, then FRAMOS are responsible for its technical support as it is not an Intel product.  So questions should be directed to FRAMOS.  Thank you.

Please visit the link below and click on the **Questions?  We are happy to help!** button in the bottom corner of the site.

https://www.framos.com/

[image]

jpenaloza1211, 2020-07-15:
Hi, I apologize, I made a mistake in the name. I researched both before purchasing and made a mistake in recalling the name. But I am using the **Intel Realsense D435**. I also fixed that in the original post.

MartyG-RealSense, 2020-07-15:
Okay, thank you very much.   I can answer your questions now.

1.  A camera will use its default configuration if a custom configuration is not defined.  You can define stream resolution, type and FPS speed with **config** statements (also sometimes written as **cfg**) that are placed before your pipeline start() instruction.  The word **config**(or cfg if that term is used instead) should also be placed in the brackets of the pipe start instruction to tell the camera to use your custom configuration.  

The opening lines of the Python example program script **opencv_viewer_example.py** provide an example use of cfg setup:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

Other kinds of settings, such as exposure time, can either be defined individually with scripting or by loading in a file called a **json** that contains a mass list of settings.  Here is an example Python script that specifies a json file to load in its opening lines:

https://github.com/IntelRealSense/librealsense/issues/6075

You can easily generate a json file by defining settings in the **RealSense Viewer** program and then exporting them as a json using a button on the toolbar at the top of the Viewer's options side-panel.

2.  A good starting point for learning about changes covered by the json file may be to try different settings in the RealSense Viewer and see how they affect the values in an exported json when you open the file with a text-editor program.

Learning about stream configuration with **config** instructions may be more straightforward.  If you know the correct term for the stream type (the 'format') then if you have a table of supported resolutions and FPS speeds then it is simple to substitute those values into the config statements.  To break down a typical pair of statements:

**config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)**

In Python, you tell the program that you want to use a depth stream with **rs.stream.depth** whilst if you want to define an RGB color stream, you specify **rs.stream.color**

Next, you split the resolution into a pair of values.  So for example, 640x480 is written as **640, 480**, and 1280x720 would be written as **1280, 720**

Then you define the stream format.  This is easy for depth as it is always **z16** - thus written as **rs.format.z16.**  For other stream types such as color and infrared there are a greater number of formats to choose from.  For infrared, **rs.format.y8** is a good default choice, whilst for color **rs.format.bgr8** should be fine as a default choice.

Finally, you provide a number to define the FPS speed, such as **30.**  This should be a value from a fixed range of available possible settings (6, 15, 30, 60, 90).  Some FPS speeds may not be available for certain resolutions.

The data sheet document for the 400 Series cameras contains a list of supported resolutions and their supported fPS speeds on page 59 onward of the current edition.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

MartyG-RealSense, 2020-07-15:
I recommend deliberately specifying that auto-exposure should be enabled.  The link below provides example Python code for enabling and disabling auto-exposure.

https://github.com/IntelRealSense/librealsense/issues/1624

Setting rs.option.enable_auto_exposure to 1 enables auto-exposure, and setting it to 0 disables it.

jpenaloza1211, 2020-07-15:
got it, thank you for all the help! I'll close the issue now since that answers everything for me.

closed 2020-07-15 by jpenaloza1211