realsenseai/librealsense #6834: The short distance to get the pointcloud

wuhaoqi-hub, 2020-07-15:
What is the shortest distance of the point cloud captured by the camera? Is there a manual to get this data? On the other hand, does the color of the object surface affect the point cloud quality? Because I found that the black surface gets better point cloud quality than the white surface. Why?

dorodnic, 2020-07-15:
At the moment we have 3 product lines - structured light SR300, stereo D400 and lidar L500.
Which camera are you using?

wuhaoqi-hub, 2020-07-16:
D435 resolution is 680*360

wuhaoqi-hub, 2020-07-16:
> At the moment we have 3 product lines - structured light SR300, stereo D400 and lidar L500.
> Which camera are you using?

D435 resolution is 680*360

RealSenseCustomerSupport, 2020-07-16:
----------------------------------------------

Hi wuhaoqi-hub

Under 640x360 on D435, the minimum depth distance is around 16cm (around 6"). Stereo 3D camera likes to see the objects with random texture patterns regardless of the color. If there is nothing on object surface, i.e. white wall, the laser projector can add IR dots on the object surface to assist camera to get 3D vision.

RealSenseCustomerSupport, 2020-07-24:
----------------------------------------------

Any further question? otherwise the ticket will be close next week. thanks

closed 2020-07-31 by RealSenseCustomerSupport