realsenseai/librealsense #6835: Get Lidar data from L515

tuencfs, 2020-07-15:
Dear Sir or Madam,
Our company is interested in using one of your L515 devices.

Currently, we use a 3D LiDAR device for our application, the data output from this LiDAR is point cloud with x,y,z, and intensity, ring index value. 

Can I use L515 like a 3-D LiDAR to our application or We only use L515 like a depth camera?

Is there any way to get direct LiDAR data (x,y,z, intensity ... ) from L515?

Thanks

dorodnic, 2020-07-15:
The SDK includes method for converting depth data into (x,y,z)
Intensity and confidence are provided as additional streams, alongside depth. 
I'm not familiar with ring index value. 
What type of interface are you using right now for interfacing with the device? 
Is it exposed as a specific ROS topics?

tuencfs, 2020-07-16:
Hi @dorodnic 
Yes, we receive data from LiDAR over ROS topics from VLP-16. So L515 is a type of depth camera like D435i right?

We have ordered D435i and L515, both of them will arrive at the end of July. Can I also get the Intensity and confidence value from D435 or D435i?

dorodnic, 2020-07-19:
Yes, L515 is very similar to the D435i from interfaces point-of-view and behaves as a depth camera. 
You can get Intensity and Confidence from the L515.
We also provide ROS driver, but the format is likely different.

zainmehdi, 2020-07-28:
> The SDK includes method for converting depth data into (x,y,z)
> Intensity and confidence are provided as additional streams, alongside depth.
> I'm not familiar with ring index value.
> What type of [...]

Hi @dorodnic , Can you please reference the relevant function / code snippet for acquiring the x y z values. And is there any possibility that the SDK will expose access to LIDAR stream in future. It would be helpful if LIDAR data could be streamed to allow for scan matching and other operations etc, Thanks

RealSenseSupport, 2020-08-13:
Hi @tuencfs 

Were you able to get the information you needed here?

Hi @zainmehdi 

We don't have any plans to provide the LiDAR stream directly.  Although will take note of it.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-08-20 by RealSenseSupport

yhecarpio, 2022-03-03:
Hi @dorodnic 
How can I get the additional intensity stream? I'm working with L515 to create point-cloud files in KITTI format, so I would like to get the intensity values along with (x, y, z) data. Is this possible?