realsenseai/librealsense #6836: Question: how to confirm that you are using RSUSB or a patched kernel module

jurc192, 2020-07-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D4000 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version  | Linux   | 
| Kernel Version (Linux Only)   |  (5.4.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.36.0                           | 
| Language                            | python                          | 
| Segment			| others }                          | 

### Issue Description

Short question: how can we verify if RealSense SDK is using RSUSB or (patched) kernel drivers?

I am trying to compare the performance between the two, compiling the realsense SDK from source on Linux.
Is there a reliable way (sanity-check) to see which of the two is currently being used- rsusb or kernel drivers?

dorodnic, 2020-07-15:
Hi @jurc192 
We've been planning dedicated API for this for a while, but for now you can look at Physical Port camera info. For V4L2 it would be `something...v4l2...` while for RSUSB it would be `1-2-3`.
With V4L2, you can check if its patched by checking timestamp domain property of a frame.

RealSenseSupport, 2020-07-17:
Hi @jurc192 Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

jurc192, 2020-07-20:
Thanks for your answer!
I can indeed see the difference by checking RealSense Viewer app -> info -> Physical port :)

closed 2020-07-20 by jurc192