realsenseai/librealsense #6839: Realsense Viewer install (windows) - per frame metadata is not enabled at the os level

Fred3D-tech, 2020-07-15:
----------------------------------------------------------------------------------------------------
Windows 10

Hello with the last version of realsense viewer, I get this message on some depth frame (using a L515):
"per frame metadata is not enabled at the os level
please follow the installation guide for the details"

The problem is I can't find any installation guide... How can I enable per frame data ?
thanks

dorodnic, 2020-07-15:
Hi,
Usually the Viewer should prompt you to enable metadata via a pop-up message, did you see / try it?

Fred3D-tech, 2020-07-15:
hi,
it was the case previously, but not in the last version...

RealSenseSupport, 2020-07-23:
@Fred3D-tech 

When you say previously, do you mean previous librealsense version with L515?

As dorodnic stated, the current viewer should ask via a pop-up prompt message.

RealSenseSupport, 2020-08-18:
@Fred3D-tech 

Were you able to come across the pop-up prompt message in the Viewer?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-08-27 by RealSenseSupport