realsenseai/librealsense #6841: Directly recording mp4, mpeg, etc video without having to record bag

matthewha123, 2020-07-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Jetson Nano  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

I'm wondering if it is possible to record video from a realsense camera without having to record to a .bag file first. I've seen that .bag files can easily be converted to .mp4's, but, since .bag files are so large, I wanted to circumvent that intermediary step.

MartyG-RealSense, 2020-07-15:
Hi @matthewha123   In this scenario, I would personally use video screen-capture software such as **XSplit Broadcaster**  to record the complete contents of the screen to mp4 video whilst letting the camera stream in real-time in a full-screen view but not recording to a bag.

https://www.xsplit.com/broadcaster
https://www.xsplit.com/broadcaster/manual/record

dorodnic, 2020-07-20:
Also, if you build the SDK with V4L2 backend (`-DFORCE_LIBUVC=OFF`) you should be able to capture raw data from the camera using standard Linux tools. It would not be nicely colorised, but it should work.

matthewha123, 2020-07-20:
Thank you for the suggestions @dorodnic and @MartyG-RealSense

closed 2020-07-20 by matthewha123