realsenseai/librealsense #6844: DeviceList#getDeviceCount()  return 0 on Android with L515

kyosho-, 2020-07-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |  Android 10 | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | Google Pixel 3a  |
| SDK Version                            |  2.36.0                          | 
| Language                            |  Java                          | 
| Segment			|  Smartphone                          | 

### Issue Description

When I run examples on Android with RealSense L515, the examples is not working.

After investigating, I found that DeviceList#getDeviceCount() method return 0.
Essentially, the method should return a value greater than 0.

Android device detect L515 and display permission dialog when I connect L515.

USB cable is suppoted OTG.
DeviceList#getDeviceCount() method return 1 using D435i.

RealSenseSupport, 2020-07-23:
Hi @kyosho- 

We are reviewing our Android support.  Will keep this ticket in mind as we review the support for Android.

kyosho-, 2020-07-24:
Thank you for reply. I hope that it will be improved quickly.

RealSenseSupport, 2020-09-04:
Hi @kyosho- 

There has been an update in the latest release with regards to Android.  Have a look at the release:

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2381

Also here is the PR that encompasses the Android updates that has been merged in:  https://github.com/IntelRealSense/librealsense/pull/6945

** Please be informed that we will close this ticket in seven (7) days if we have not heard back from you. You're encouraged to close the ticket if you think the issue/problem is solved.

kyosho-, 2020-09-12:
Thank you for reply.

Unfortunately I don't have L515 now, so I can't verify.
I will verify and report when it is available.

closed 2020-09-12 by kyosho-