realsenseai/librealsense #6845: D415 RGB sensor issues

Erandurthil, 2020-07-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.6.0 | 
| Operating System & Version |  Debian  | 
| Kernel Version (Linux Only)    |  5.3  | 
| Platform                                 | PC |
| SDK Version                            | 2 | 
| Language                            |  C++  | 

### Issue Description

I discovered that both my D415's, one being new out of the box, have the same issue under Linux but not under Windows. 
They show a single, partial line of green pixel's in the bottom right corner in both the Realsense-Viewer 2D streams, and the examples provided in the SDK. However those green pixels don't appear in the 3D view of the Realsense-Viewer.

Using the same cameras and the same examples, this issue did not occur on windows. 
[This person](https://support.intelrealsense.com/hc/en-us/community/posts/360048022393-Dead-green-pixels-in-stream-) has the same exact pattern with a different camera model and reported it about two weeks ago, so I don't think this is an calibration or hardware issue, since both on-chip and dynamic calibration did not alleviate the issue. 

### Linux Screenshots

Realsense Viewer 3D view (Linux):
[image]

Pointcloud Example (Linux):
[image]

Multicam Example (Linux):
[image]

### Windwos Screenshots

Multicam Example(Windows):
[image]

Pointcloud Example(Windows):
[image]

MartyG-RealSense, 2020-07-16:
Would it be possible to do a build of librealsense using the **backend** installation method please?  It requires an internet connection but is not dependent on Linux versions or kernel versions and does not need patching, as dependencies are installed over the internet.  This should help to deal with any problems that are caused by patching or kernel related issues.

The first step if not already done is to download a source-code zip file for SDK version **2.36.0** from the "Assets" list at the bottom of the release page information:

https://github.com/IntelRealSense/librealsense/releases/tag/v2.36.0

[image]

Then please follow the installation instructions for the **backend** method in the link below:

https://github.com/IntelRealSense/librealsense/issues/6710#issuecomment-650611442

Erandurthil, 2020-07-17:
Thanks, that solved the issue !

closed 2020-07-17 by Erandurthil

MartyG-RealSense, 2020-07-17:
Great news - thanks so much for the update!