realsenseai/librealsense #6846: Intel Realsense viewer does not recognize Intel realsense D415 camera.

SkarsgardSaga, 2020-07-16:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Linux|

Please provide a description of the problem

MartyG-RealSense, 2020-07-16:
Hi @SkarsgardSaga   What information do you receive when doing this command please:

**modinfo uvcvideo | grep ver**

SkarsgardSaga, 2020-07-16:
> Hi @SkarsgardSaga What information do you receive when doing this command please:
> 
> **modinfo uvcvideo | grep ver**

I get:

version:          1.1.2.realsense-1.3.13
description:    USB Video Class driver
srcversion:     3DD05991EAABF8F5DE4E188
vermagic:       5.3.0-59-generic SMP mod_unload

MartyG-RealSense, 2020-07-16:
The uvc kernel module information provided by that command looks okay to me.  Could you try doing a complete shutdown of your computer (not a restart) and then boot up with the power button and see if the camera is detected in the Viewer?

SkarsgardSaga, 2020-07-16:
I did it, but I still get the same result as before

SkarsgardSaga, 2020-07-16:
I got an update in the meantime and I see that my version has changed to:

version:          1.1.2.realsense-1.3.14
description:    USB Video Class driver
srcversion:     32DE67CAA7876861B634FAE
vermagic:       5.3.0-62-generic SMP mod_unload

MartyG-RealSense, 2020-07-16:
What computer are you using the camera on, please?  (PC, Mac, Jetson, Raspberry Pi, etc)

SkarsgardSaga, 2020-07-16:
I am using a Lenovo ThinkPad

SkarsgardSaga, 2020-07-16:
Uninstalled everything and reinstalled it, but still ended up with the same result. Works just fine on my colleague's Dell computer however.

MartyG-RealSense, 2020-07-16:
If the same camera works fine on your colleague's computer then that would make it unlikely to be a firmware driver related issue.  This is because the firmware is stored inside the camera hardware and not on the computer, so any problem with the firmware would be carried over to other computers that the camera was attached to.

Is you r colleague Dell running Linux too like yours or is it a Windows PC?

SkarsgardSaga, 2020-07-16:
Thanks for being so helpful.
yes, both my colleague and myself have Windows and Ubuntu partitioned computers and we both used ubuntu when testing the camera.

MartyG-RealSense, 2020-07-16:
If the camera is working in Ubuntu on your colleague's computer then it is unlikely to be related to the dual-boot setup of the computers.  If you installed librealsense in exactly the same way on both computers, and they are not networked (they can have different software on them), that raises the possibility that they have Linux driver versions of different ages.

Is it possible to update the graphics drivers on your particular PC?

Also, is it possible for you to use the Windows version of the RealSense SDK?

pirate999, 2020-07-20:
@SkarsgardSaga Have you solve the problem? I meet the same problem as yours.

pirate999, 2020-07-27:
Ok,thank you!

MartyG-RealSense, 2020-08-08:
@SkarsgardSaga   Do you require further support with this case please, or can it be closed?   Thanks!