realsenseai/librealsense #6847: [Python] How to get get_distance() to work on a depth image?

kadhirumasankar, 2020-07-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Gazebo  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                          | 
| Segment			|  Drone                          | 

### Issue Description
Hello, sorry if this question has already been asked, but I haven't been able to find anything yet. I have a drone in a Gazebo simulation with the D435 on it. In a Python file, I have a subscriber to `/camera/depth/image_raw`. In the callback, I used `imgmsg_to_cv2` to convert it to an image. How do I get the `get_distance()` function to work on this image? I've pasted my code below:
```python
#!/usr/bin/env python2

import rospy
from cv_bridge import CvBridge, CvBridgeError
from sensor_msgs.msg import PointCloud2 as sensor_msgs_pc2
import sensor_msgs.point_cloud2 as pc2
from sensor_msgs.msg import Image
import pyrealsense2 as rs
import cv2

class CameraFeed(object):

    def __init__(self):
        print('[INFO] Initialized object')
        self.image_sub = rospy.Subscriber("/camera/depth/image_raw",
                                          Image,
                                          self.depth_camera_callback)
        self.bridge_object = CvBridge()

    def depth_camera_callback(self, data):
        try:
            original_image = self.bridge_object.imgmsg_to_cv2(
                data,
                desired_encoding="passthrough"
            )  # bgr8 is the default OpenCV encoding
        except CvBridgeError as e:
            print(e)
        print('[INFO] Got callback')
        cv2.imshow('Image', original_image)
        cv2.waitKey(3)
        rs.depth_frame.get_distance()
        raw_input()
        print('[INFO] Finished callback')

def main():
    print('[INFO] Running file')
    camera_feed_object = CameraFeed()
    rospy.init_node("camera_feed_node", anonymous=True)
    try:
        rospy.spin()
    except KeyboardInterrupt:
        print("Shutting down")

if __name__ == '__main__':
    main()
```

Any kind of help would be greatly appreciated!

FLS-BP-US, 2020-07-17:
You can output the image to a text file with:
np.savetxt(filepath, depth_image, fmt='%d', newline='\n', delimiter=' ')
Each pixel will have a value of depthunits. (default for d435 is 0.001m and the L515 is 0.00025m)

MartyG-RealSense, 2020-07-19:
Hi @kadhirumasankar   The approach that I usually see is to set up a 'depth' variable whose value is the depth value multiplied by the depth scale, and then calculate distance this way:

**dist,_,_,_ = cv2.mean(depth)**

Intel's Python **distance_to_object tutorial** for displaying the depth of an object on-screen uses this method.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

Another way in which this distance calculation technique is used by a RealSense community member in a simpler form is here:

https://github.com/IntelRealSense/librealsense/issues/3473

kadhirumasankar, 2020-07-20:
Hi @MartyG-RealSense, thanks for your reply. I had seen that tutorial before, but I couldn't use it because I can't set up a pipeline since my RealSense camera is on a drone in a Gazebo simulation environment. I have to subscribe to ROS topics and use their callbacks. I found out that subscribing to the `/camera/depth/image_raw` topic gave me an image which I could index with my coordinates of interest, and it seems to output the depth in mm. I verified this by seeing what the max range of the camera was set to and the output cutoff at the same number.

Is that correct? Does indexing the image indeed give you the depth in mm? If so, I'll close the issue once I get your reply

MartyG-RealSense, 2020-07-20:
According to Doronhi the RealSense ROS wrapper developer, depth is in the ROS standard scale of mm.

https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873

kadhirumasankar, 2020-07-20:
Great, looking like just indexing the depth image works then. Thank you!