realsenseai/librealsense #6850: T265 can't be used: RS2_USB_STATUS_BUSY

tpwrules, 2020-07-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |   Lubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4.0                                   | 
| Platform                                 | PC |
| SDK Version                            |  2.36.0 | 
| Language                            |  N/A                          | 
| Segment			|  {VR}                          | 

### Issue Description
I've just compiled librealsense with `-DFORCE_RSUSB_BACKEND=ON` because I can't recompile the kernel for various reasons. Unfortunately, my T265 does not work (though my D435 works great).

The output of `rs-enumerate-devices -d` is shown below with just the T265 plugged in. I've tried multiple USB ports, cables, powered and unpowered hubs, etc., and no change. I've verified the udev rules are appropriately installed.

```$ rs-enumerate-devices -d
Device info: 
    Name                          :     Intel RealSense T265
    Serial Number                 :     9<redacted>
    Firmware Version              :     0.2.0.951
    Physical Port                 :     2-2-14
    Product Id                    :     0B37
    Usb Type Descriptor           :     3.1
    Product Line                  :     T200

Default streams:
 16/07 15:48:15,823 ERROR [140403953100672] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 16/07 15:48:15,823 ERROR [140403953100672] (types.h:308) Unable to open device interface
2020-07-16 15:48:15,823 WARNING [default] Could not open device Unable to open device interface
 16/07 15:48:15,825 ERROR [140403953100672] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 16/07 15:48:15,825 ERROR [140403953100672] (types.h:308) Unable to open device interface
2020-07-16 15:48:15,825 WARNING [default] Could not open device Unable to open device interface
 16/07 15:48:15,826 ERROR [140403953100672] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 16/07 15:48:15,826 ERROR [140403953100672] (types.h:308) Unable to open device interface
2020-07-16 15:48:15,826 WARNING [default] Could not open device Unable to open device interface
RealSense error calling rs2_config_resolve(config:0x1e97110, pipe:0x1eb1080):
    No device connected
```

Attempting to start a pipe with pyrealsense2 shows the message `2020-07-16 15:49:45,342 WARNING [default] Cloud not open device Unable to open device interface` repeatedly and no frames come through.

tpwrules, 2020-07-17:
Closing as I've isolated the bug significantly.

closed 2020-07-17 by tpwrules

drapado, 2020-09-22:
Hi @tpwrules, did you manage to solve the issue? if so, how? Thanks!