realsenseai/librealsense #6851: calling rs2_pipeline_start_with_config(pipe:0x28be1f80, config:0x28b57830):     Cannot open '/dev/video10 Last Error: No such file or directory

hhhh-1111, 2020-07-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |  linux/buildroot  | 
| Platform                                 | arm  |
| Language                            |  C++       | 

### Issue Description
code:
```
rs2::config cfg;
rs2::pipeline pipe;
rs2::pipeline_profile profile;
rs2::frameset data;

try {
    profile = pipe.start(intelCamera.cfg);
    data = pipe.wait_for_frames(); 
    auto depth_frame = data.get_depth_frame();
    pipe.stop();
} catch (const rs2::error & e) {
    cerr<<"RealSense error calling "
            <<e.get_failed_function()
            << "(" << e.get_failed_args() << "):\n    "
            <<e.what()
            <<endl;
} catch (const exception & e) {
    cerr<<e.what()<<endl;
}
```
The following errors occur occasionally:
```
RealSense error calling rs2_pipeline_start_with_config(pipe:0x28be1f80, config:0x28b57830):
    Cannot open '/dev/video10 Last Error: No such file or directory
```

MartyG-RealSense, 2020-07-17:
The error may be related to a similar one in the case in the link below, which may have been caused "when the application tries to configure stream with unsupported resolution/fps/format modes".

https://github.com/IntelRealSense/librealsense/issues/4931

Looking at your script, I note that you are calling a custom **cfg** configuration in your pipeline start but do not seem to have defined any cfg instructions before that.  I would also expect that the call to cfg in the brackets of your pipe.start instruction would use the same term that was defined in **rs2::config cfg**.  So instead of:

** profile = pipe.start(intelCamera.cfg);**

I would recommend using this:

** profile = pipe.start(cfg);**

hhhh-1111, 2020-07-17:
code:
```
rs2::config cfg;
rs2::pipeline pipe;
rs2::pipeline_profile profile;
rs2::frameset data;

cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

while(true) {
    try {
        profile = pipe.start(cfg);
        data = pipe.wait_for_frames(); 
        auto depth_frame = data.get_depth_frame();
        pipe.stop();
    } catch (const rs2::error & e) {
        cerr<<"RealSense error calling "
                <<e.get_failed_function()
                << "(" << e.get_failed_args() << "):\n    "
                <<e.what()
                <<endl;
    } catch (const exception & e) {
        cerr<<e.what()<<endl;
    }
    sleep(10);
}
```
Most of the time it's OK．
Only occasionally, the errors occur．
```
RealSense error calling rs2_pipeline_start_with_config(pipe:0x28be1f80, config:0x28b57830):
    Cannot open '/dev/video10 Last Error: No such file or directory
```

MartyG-RealSense, 2020-07-17:
Does it still work if you remove the **sleep(10)** instruction?  Whilst it is important to put the CPU to sleep for a time when using **poll_for_frames()** to stop the CPU usage from maxing out, it should not matter so much with **wait_for_frames()**.

MartyG-RealSense, 2020-08-02:
Hi @hhhh-1111    Do you require further assistance with this case please, or can it be closed?   Thanks!

closed 2020-08-03 by hhhh-1111