realsenseai/librealsense #6854: Question: is it possible to reduce building time for wrapper examples?

milan-r-shah, 2020-07-17:
| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                    | 
| Firmware Version                   | 05.12.05.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.3.0-62-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0                          | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

### Issue Description
I followed [Linux Ubuntu Installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) guide for building **librealsense2** from scratch.
As I wanted to work with examples, I had run
```
cmake ../ -DBUILD_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make install 
```

which took considerable amount of time. 

---

Later on, I wanted to work on PCL, so I followed [PCL wrapper documentation](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl). For the PCL wrapper examples, in the `<librealsense root dir>/build` directory, I ran:
```
cmake -DBUILD_PCL_EXAMPLES=true ..
sudo make uninstall && make clean && make && sudo make install
```
which also took quite a long time.

---

So, my question is once the SDK has already been built (along with standard examples), are there any ways to reduce the building time for different wrapper examples (apart from _`make -jX`_) ? In the above case of `PCL`, have I run any redundant commands/steps?

Because if I want to build the examples for other wrappers e.g. `OpenCV` then I have to follow similar steps, which would again take quite a long time, right? 

Thanks a lot in advance :)

MartyG-RealSense, 2020-07-18:
As far as I know, using -jx to take advantage of core numbers is the main official way to speed build time with librealsense.   I would speculate (it is not something I have tested) that you may be able to get further optimisations if instead of trying to accelerate the compilation of librealsense specifically, you focus on improving CMake's general compilation speed.  Googling for **increase cmake compilation speed** can provide a wide range of suggestions.  Also, use a recent version of CMake to take advantage of any speed optimisations that may have been added to it since earlier versions.

MartyG-RealSense, 2020-08-02:
Hi @milan-r-shah   Do you require further assistance with this case, or can it be closed now?   Thanks!

milan-r-shah, 2020-08-04:
Hi @MartyG-RealSense ,
Sincere apology for the late response. 
Thank you for all the suggestions. However, I want to confirm a few things:
1. Currently, I can't build just only one example e.g. **hello-realsense**, right? First, I have to build all of them and then if I make changes to any particular example, I can run `make` again for that specific one, right?
2. Let's say if I want to build examples as well as pcl-wrapper examples then can I do something this?
```
cmake -DBUILD_EXAMPLES=true -DBUILD_PCL_EXAMPLES=true ..
sudo make uninstall && make clean && make && sudo make install 
```
3. If I have already built

milan-r-shah, 2020-08-04:
Hi @MartyG-RealSense ,
Sincere apology for the late response. 
Thank you for all the suggestions. However, I want to confirm a few more things:

1. Currently, I can't build just only one example e.g. **hello-realsense**, right? First, I have to build all of them, and then if I make changes to any particular example, I can run `make` again for that specific one, right?

2. Can I simultaneously build general examples as well as pcl-wrapper examples like this?
```
cmake -DBUILD_EXAMPLES=true -DBUILD_PCL_EXAMPLES=true ..
sudo make uninstall && make clean && make && sudo make install 
```
3. Let's say if I have already built general examples using this command:
```
cmake -DBUILD_EXAMPLES=true ..
sudo make uninstall && make clean && make && sudo make install 
```
and later on, if I want to build pcl-wrapper examples then removing 'sudo make uninstall` and running
```
cmake -DBUILD_PCL_EXAMPLES=true ..
make clean && make && sudo make install
```
would save time?

MartyG-RealSense, 2020-08-04:
1.  My understanding is that you build all of the examples and then run **make** after making changes to one, and only the ones that have changed are updated.

2.  You should be able to use as many customization flags as you want to build additional features at the same time as building the core librealsense.  Adding all the flags at once does create complication in diagnosing which one is causing a conflict if the build does not work though.  The build log should give clues about where the problem is coming from.

3.  You would usually put **../** after cmake and then put your flag after that.  for example:

cmake ../  -DBUILD_PCL_EXAMPLES=true

milan-r-shah, 2020-08-05:
Hi @MartyG-RealSense ,
Thanks for the confirmation on 1 & 2.

My 3rd question is a little bit different than what you have mentioned in your answer:
In that one, I was saying that let's say I have already built the general examples using `sudo make install` and later on if I want to build pcl-wrapper examples then not running `sudo make uninstall` command, would make any difference, from the build-time point of view?

MartyG-RealSense, 2020-08-05:
On 3.  I think I would follow the official instructions and keep **uninstall** in, but use **make -j8** to accelerate build time by allowing multiple cores in the computer's processor to be used during building.  This may be best used on a desktop or laptop computer, as the documentation advises that using the -jx command on low-end systems can cause the build to randomly hang.

**sudo make uninstall && make clean && make -j8 && sudo make install**