realsenseai/librealsense #6856: Invert colors in depth image

Borjolf, 2020-07-18:
Hi,

Is there a way to invert colors in a depth image?
When displaying this:
 depth_image = np.asanyarray(depth_frame.get_data()) * depth_scale

I get a depth image where closer points to the camera are darker, but I want it to be exactly the opposite, without losing the distance information ( the closer the point the lighter).

Any idea? Thanks

MartyG-RealSense, 2020-07-18:
There was a case in March 2020 where a RealSense user programming in Python was getting depth displayed as red in the near range and blue in the far range when they applied a colorizer to the depth data.  The approach taken to invert the colors was to load a different **visual preset** file.

https://github.com/IntelRealSense/librealsense/issues/3363

The case in the link above contains a link directing to a list of number codes for the visual presets.  The link below contains the correct jump-to point for the list.

https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_processing.hpp#L718

Borjolf, 2020-07-21:
Thanks! This worked fine.

closed 2020-07-21 by Borjolf

MartyG-RealSense, 2020-07-21:
Great news - thanks for the update!