realsenseai/librealsense #6859: L515 not showing up in Unity

VisibleMeasurement, 2020-07-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515} | 
| Firmware Version                   | (1.4.2.1) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | Unity  |
| SDK Version                            |  { 2.36 }                          | 
| Language                            |  {unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<I am getting the "Start Here" scene to open in Unity 2018, but the L515 will not show up once the Scene is in Play mode.  This happens in other provided scenes as well.  I get the message "Connect Sensor" but I have the L515 plugged into the computer on the provided USB3 to USBC cable.

I can see the L515 in the RealSense viewer 2.36 using firmware 1.4.2.1 so it is working otherwise.  Is the sensor not recognized by Unity yet?  

Thanks for any help in advanced.  I'm looking to do some trajectory mapping with the device and SLAM scanning.  If anyone can guide me to a better method than Unity, for a non-programmer, let me know.  I'm wanting to access the trajectory and associate it with other sensors plugged into the laptop.>

RealSenseSupport, 2020-07-23:
Hi @VisibleMeasurement 

Were you able to try the PR mentioned by dorodnic?

VisibleMeasurement, 2020-07-25:
I pulled down the development branch and tried to open the Unity scene provided using the new download.  However, it still doesn't recognize the L515.  I did notice that I am having to load the same unitypackage file as before, so that might be a problem.

I am teaching myself how to compile the code, but it's taking me a while, I've never done that before.  I'm a power PC user but not a programmer.  I'll let you know if I have any success.

dorodnic, 2020-07-27:
Hi @VisibleMeasurement 
Please try the Unity Package from [v2.37.0 pre-release](https://github.com/IntelRealSense/librealsense/releases/tag/v2.37.0)

chrisrogers3d, 2020-08-02:
Hello, Ive installed the 2.37 prerelease sdk and 2.37pr unity package.  My 515L has 1.4.2.1 firmware.  Works with realsense viewer 2.37.  Unity 2019.3.13 Windows10.  Also tried to build from source but ran into atlcom.h missing headers, reinstalled visual studio 2019 but same problem.

In unity when running the DepthAndColor scene (that works with my D455) the error message when trying to connect either with or without "requested serial number" is:

ExternalException: rs2_pipeline_start_with_config(pipe:000002062D592D40, config:000002062D592BE0)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <782525f085c64debb7028293360b3d39>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <782525f085c64debb7028293360b3d39>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

VisibleMeasurement, 2020-08-11:
It took a while to get to it, but I did get the camera to work with the default scene and the new unity package.  It mimics the function of the realsense viewer.

RealSenseSupport, 2020-09-11:
Now that the Unity package was merged, is there anything else that is needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

VisibleMeasurement, 2020-09-16:
This is closed for me, thank you for the help.

closed 2020-09-16 by RealSenseSupport

AndreV84, 2020-10-27:
is L515 supported as for today?