realsenseai/librealsense #6860: Motion Module Failure for D435i

ashirsat, 2020-07-19:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|841612070960|
|**Firmware Version**|05.12.06.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|903223051743|
|**Firmware Update Id**|903223051743|

Linux Kernel: 5.3.0-62-generic

I get a error for when I turn on the motion module for relasense d435i with the message motion module failure in Intel Realsense viewer.

Is this a hardware issue ?

Regards,
Aniket

MartyG-RealSense, 2020-07-19:
Hi @ashirsat   The link below describes some RealSense users' past experiences of this error.  My understanding is that it does not prevent the motion module from working fine in the Viewer when it occurs, though it may have consequences for ROS-based applications.

https://github.com/IntelRealSense/librealsense/issues/6008

ashirsat, 2020-07-21:
@MartyG-RealSense  Yes I wanted the imu readings for my application. I will check if it affects the imu stream in ros and keep you updated.

MartyG-RealSense, 2020-08-02:
Hi agan @ashirsat  Do you still require assistance with this case please, or can it be closed?   Thanks!

ashirsat, 2020-08-02:
Hi @MartyG-RealSense , 

I did see the imu readings been published on the combined topic using the interpolation method mentioned in the README.md. I wanted to know why that error message comes up even though the stream from the imu (gyro + accel) is actively sending data.

Thanks,

MartyG-RealSense, 2020-08-03:
The motion module failure warning pop-up can be considered as a glitch in the activation process of the IMU streaming.  I'm not aware of a precise known cause for it, but it is usually harmless.  It is not unusual for a one-off error to sometimes be generated when a process starts, has a momentary hiccup and then recovers itself.

In case you plan to use RGB and IMU together with 2.36.0, it is also worth mentioning that there is a bug in 2.36.0 in the RealSense Viewer only (not any other part of the RealSense SDK) that can disrupt RGB when the motion module is activated.  The link below contains details of workarounds for this.

https://github.com/IntelRealSense/librealsense/issues/6992#issuecomment-667633901

closed 2020-08-16 by MartyG-RealSense