realsenseai/librealsense #6861: Question: How to return high speed pose data and 30fps image data from T265, without callback?

antithing, 2020-07-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {2.4 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

When using the callback API, as in the pose_and_image example, I find that the tracking latency of the T265 rises after an hour or so of running, from very low (6ms) to very high (800+ ms).

When I do not use the callback, and instead use:

```
while (exitrequest == false) {

		{
			// Wait for the next set of frames from the camera
			auto frames = pipe.wait_for_frames();
			// Get a frame from the fisheye stream
			rs2::video_frame fs = frames.get_fisheye_frame(fisheye_sensor_idx);
			// Get a frame from the pose stream
			rs2::pose_frame fp = frames.get_pose_frame();

			auto pose_data = fp.get_pose_data();
        }

      }

```

I do not see this issue, but the pose data comes in at 30 fps, with the image frames.

How can i return the high speed pose data and the slow speed image data without using a callback?

Thanks!

dorodnic, 2020-07-20:
Using the callback is the way to retrieve high speed data.
If asynchronous API is inconvenient, you can use `rs2::frame_queue` - in the callback enqueue any incoming frames, in your main thread do `wait_for_frame` or `poll_for_frame` on the queue. You still have to use callbacks, but it simplifies the code significantly. 
As for latency in rs-trajectory, could it be due to `trajectory` vector filling-up much faster with callbacks and hence taking longer to draw?

antithing, 2020-07-20:
Thanks for getting back to me.

I have tried using `wait_for_frame `, but that seems to pull frames at 30fps only. Can i use this to get the 200fps tracking data?

I am not using the trajectory  vector, just the same callback code as the example.

closed 2026-10-23 by MartyG-RealSense