realsenseai/librealsense #6866: Official Support for Linux Kernel 5.4+

YoshuaNava, 2020-07-20:
### Issue Description
I have D435 and D435i cameras which I use on a robotic system. I make heavy use of the cameras' metadata to compute the frame timestamps and monitor the status of the sensor while the robot operates.

At the moment I'm considering updating my OS from Ubuntu 18.04 LTS with Kernel 4.15.0-48, to Ubuntu 20.04 with Kernel 5.4+ (starting on 5.4.0-40). In both cases, I aim to use the `lowlatency` version of the kernel.

I tested the librealsense software on the aforementioned OS candidate, and found that the dkms module from librealsense is currently not able to patch the required modules while working with it. I delved deeper into the scripts provided in this repo, and after modifying them a bit I was able to patch the UVC Kernel module with the setup mentioned above.

As of now, I can proceed with running the scripts to patch my kernel, but long-term I would like to be able to fall back to the DKMS solution, and also fix the kernel version I use on the robots to a specific one. Thus I would like to ask the developers of the librealsense repo:

1. By when is it planned to offer official support for Linux Kernel 5.4+ and Ubuntu 20.04+?
2. Do you plan to support a particular version of the kernel and OS?
3. I noticed that the manual kernel patching introduces an additional dependency: the *HWE* stack. Why is this needed, and will it be needed long-term?

I am aware that there are other issues regarding this, but they were closed before official support was released.

Thank you very much.

Best regards,
Yoshua Nava

ev-mp, 2020-07-21:
Hi @YoshuaNava,
1. Support for Ubuntu20 is prioritized. I cannot provide the exact schedule though but it will be delivered in weeks after kernel 5.4.
As for kernel 5.4 - it is work in progress so using the scripts from `master` branch you can already patch Ubuntu18/K5.4.
2. Support for the patches is provided for [Ubuntu with LTS kernels](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages).
3. Can you explain what do you mean by "HWE Stack" dependency ? The SDK dependencies are separate from the kernel, and for the DKMS only the standard linux tools are required (e.g. `gcc`).

>...In both cases, I aim to use the `lowlatency` version of the kernel

The DKMS packages are built, tested and distributed with `x.y.z-generic` LTS kernels. And while the differences should be minor, especially for USB drivers, deployment of DKMS on such kernel should be checked.  If I understand correctly you have already deployed DKMS on `lowlatency` kernels with Ubuntu18 PCs so (a) this apparently confirms the compatibility and (b) we would really like to get some feedback on whether you've experienced any issues with installation/usage.

YoshuaNava, 2020-07-24:
Hi @ev-mp,
Thank you for your prompt reply.

Regarding 1 and 2, I understand. I'll stay in synch with the developments in this repo. Until then I'm using a modified version of the current scripts, where I replaced 'bionic' for focal, so that the scripts could branch into the kernel 5.4 procedure while running. I've found no issues in the past with patching UVC on top of the lowlatency kernel, quite the opposite.

> Can you explain what do you mean by "HWE Stack" dependency ? The SDK dependencies are separate from the kernel, and for the DKMS only the standard linux tools are required (e.g. gcc).

The current version of the 5.4 kernel patching scripts relies on the HWE stack, as far as I know. I'm basing my judgement on this line: https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-utils.sh#L93

Thank you very much, and I remain attentive to updates from the repo.

Best,
Yoshua Nava

RealSenseSupport, 2020-07-28:
@YoshuaNava, thank you for following up closely with our development as well as all the valuable inputs. 
Could this issue be closed for now or would you like further assistance? Thank you.

YoshuaNava, 2020-07-28:
@RealSenseSupport Yes, thank you very much.

closed 2020-07-28 by RealSenseSupport

YoshuaNava, 2020-09-24:
Dear @RealSenseSupport and @ev-mp 

Do you have any news about support for Linux Kernel 5.4+ and Ubuntu Focal?

Thank you very much in advance.

doisyg, 2020-09-30:
Yes, this is much needed

doisyg, 2020-10-17:
> I delved deeper into the scripts provided in this repo, and after modifying them a bit I was able to patch the UVC Kernel module of Ubuntu 20.04/LK 5.4.0-40. I verified this by checking whether I could stream metadata through realsense-viewer.

@YoshuaNava would you be okay to share your modified script to patch the UVC kernel module of ubuntu 20.04/5.4.* ? Or even PR it ?

YoshuaNava, 2020-11-03:
@doisyg Yes. Will do asap.

@MartyG-RealSense Is there any update regarding official support (DKMS) for Linux 5.4+?

MartyG-RealSense, 2020-11-03:
@YoshuaNava   I have no information about official support for 5.4 yet.

YoshuaNava, 2020-11-18:
@doisyg Sorry for the delay. PR: https://github.com/IntelRealSense/librealsense/pull/7809

MartyG-RealSense, 2020-11-24:
Hi everyone,

Official kernel 5.4 support was added in SDK 2.40.0, along with alpha-level support for Ubuntu 20 (Focal).