realsenseai/librealsense #6867: we use cv2.VideoCapture(1) open D415, which can not work on Jetson Xavier NX

sunbin1357, 2020-07-20:
we use realsense-viewer can open D415 camera very well, but we use cv2.VideoCapture(1) to open D415, which can not work well on Jetson Xavier NX. And the screen is green.

MartyG-RealSense, 2020-07-20:
Hi @sunbin1357   My research suggests that having a green screen when using cv2.VideoCapture is not a problem that is exclusive to  The link below is an example case of this phenomenon from the Nvidia forum:

https://forums.developer.nvidia.com/t/how-to-slove-opencv-green-screen/82803

Aside from the green screen, what other problems are you having with using the D415 on an Xavier NX, please?

sunbin1357, 2020-07-21:
when I use your suggested method, the screen is not green, but is the
following result.
[image: image.png]

MartyG-RealSense, 2020-07-21:
Can you check that the gstreamer source is valid please?

https://forums.developer.nvidia.com/t/opencv-video-capture-with-gstreamer-doesnt-work-on-ros-melodic/113147/2

sunbin1357, 2020-07-21:
[image: image.png]

MartyG-RealSense, 2020-07-21:
There is a gstreamer plugin for RealSense 400 Series cameras.  It may be worth looking at if you are having problems.

https://github.com/WKDSMRT/realsense-gstreamer

sunbin1357, 2020-07-22:
Can you provide a way to call the RealSense camera via OpenCV in python？

MartyG-RealSense, 2020-07-22:
The librealsense Python wrapper's examples page has some OpenCV examples.  If you do a find search in the browser for **opencv** on that page then it will highlight all the examples that useOpenCV.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

closed 2020-08-08 by MartyG-RealSense