realsenseai/librealsense #6868: Best practice for D415 filters performance

LorenzoMonti, 2020-07-20:
Hello,

I compile from source Librealsense keeping in mind the posts about performance with temporal and spatial filters. I'm working on Nvidia AGX Xavier and i would like to exploit CUDA. I compiled with the right flags (i think) such as -DBUILD_EXAMPLES=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CUDA=true -DUSE_CUDA=true -DBUILD_WITH_OPENMP=true and i tried to remove auto-exposure.

Even though I did all this, the filters kill the performance in my project. Any suggestions?

Thanks

MartyG-RealSense, 2020-07-20:
Hi @LorenzoMonti   Which Librealsense version are you using, please?  If it is **2.35.2**, that SDK had a bug in using **-DBUILD_WITH_CUDA=true** that was fixed in SDK **2.36.0**.  

https://github.com/IntelRealSense/librealsense/pull/6581

LorenzoMonti, 2020-07-21:
Hi @MartyG-RealSense, sorry i forgot the version. The SDK version used is 2.36.0 and i've already compiled with the flag **-DBUILD_WITH_CUDA=true**

MartyG-RealSense, 2020-07-21:
CUDA is more likely to accelerate **align** functions than it is to speed up post-processing filter calculations.  

https://github.com/IntelRealSense/librealsense/pull/2670

I wonder if you may gain performance if  a ****GLSL Processing Block** is added to your application.  Like CUDA it can offload work from the CPU to the GPU.  As it can give general overall performance improvement in the SDK, you may make some speed gains that way.

https://github.com/IntelRealSense/librealsense/pull/3654

MartyG-RealSense, 2020-08-02:
Hi again @LorenzoMonti   Do you still require assistance with this case please, or can it be closed?   Thanks!

LorenzoMonti, 2020-08-04:
Thanks @MartyG-RealSense, it's all right. I'll close the issue.

closed 2020-08-04 by LorenzoMonti

MartyG-RealSense, 2020-08-04:
Thanks very much @LorenzoMonti !