realsenseai/librealsense #6870: Getting Camera Temperature D435 C#

brettmilinski, 2020-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.35.2                          | 
| Language                            |  C#                         | 

### Issue Description
How do I retrieve the Asic and Projector temperatures of the D435i camera? 
There are some C++ examples out there, but they don't seem to translate over, unless I missed something.

I see that there are the following values available in C#:
```C#
Intel.RealSense.Option.AsicTemperature
Intel.RealSense.Option.ProjectorTemperature
FrameMetadataVlue.Temperature
```

However, I have not been able to find a way to access them for a specific device.

Here's what I have:
```C#
using (var ctx = new Context())
 {
     var devices = ctx.QueryDevices();
     foreach (var device in devices)
     {
             string sn = device.Info[CameraInfo.SerialNumber];
             string name = device.Info[CameraInfo.Name];            
             names.Add(name);
             var cfg = new Config();
             if (name.Contains("D435"))
             {                         
                  cfg.EnableStream(Stream.Depth, 640, 480);                
                  cfg.EnableDevice(sn);
                  var pipe = new Pipeline(ctx);
                  pipe.Start(cfg);
                  pipes.Add(pipe);
             }
     }
}

for (int i = 0; i < pipes.Count; i++)
{    
   if (names[i].Contains("D435"))
   {      
      using (var frames = pipes[i].WaitForFrames())
       {
           using (var depth = frames.DepthFrame)
            {
                depth.GetFrameMetadata(FrameMetadataValue.Temperature);
                asicTemp = Math.Round((Convert.ToDouble(Option.AsicTemperature) * 9.0 / 5.0) + 32.0, 1);
                projectorTemp = Math.Round((Convert.ToDouble(Option.ProjectorTemperature) * 9.0 / 5.0) + 32.0, 1); 
             }
        }
    }
}
```

Is there a way for me to go through each of those devices and pull out the respective temperature options?

dorodnic, 2020-07-20:
Hi @brettmilinski 
Please try:
```cs
asicTemp = depth.Sensor.Options[Option.AsicTemperature]; // Add relevant calculations around the value
projectorTemp = depth.Sensor.Options[Option.ProjectorTemperature];
```

brettmilinski, 2020-07-20:
@dorodnic Thank you for the quick reply, that worked perfectly!

closed 2020-07-20 by brettmilinski