realsenseai/librealsense #6871: Recording to File and Supported Sensor Options

matthewha123, 2020-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  05.12.05.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | 4.9.140-tegra |
| SDK Version                            |  2.35.2                         | 
| Language                            |  python3                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

I'm interfacing with my realsense via the pyrealsense2 wrapper.
I am using this code to enable recording to a bag file and to configure some sensor options, like exposure, etc.

```
...
config.enable_record_to_file(file.bag)
profile = pipeline.start(config=config)
depth_sensor = profile.get_device().first_depth_sensor()

...

depth_sensor.set_option(<option>, <val>)
```
The code runs fine, but what concerns me is `depth_sensor.get_supported_options()` returns an empty list when I call `config.enable_record_to_file(file.bag)`. Without configuring the recording,  `depth_sensor.get_supported_options()` returns non empty list of options.

MartyG-RealSense, 2020-07-21:
Can you try changing your config statement in the pipeline start instruction's brackets please, in case doing **config = config** is negatively affecting the starting of the pipeline (the term used to define config once needs to be written in the brackets once).

**profile = pipeline.start(config)**

matthewha123, 2020-07-21:
The pipeline is able to produce frames when I use supply config as a keyword argument and when I don't (writing `config=config` vs just writing `config`). 
In either case, the `get_supported_options` still returns an empty list when called after enabling recording from a file.

MartyG-RealSense, 2020-07-21:
Apologies for the delay in responding further, as I was carefully researching your problem.  I found a case where another RealSense user had the same problem of the devices not being listed in Python.  Some suggestions for fixes are provided in this discussion:

https://github.com/IntelRealSense/librealsense/issues/2332

MartyG-RealSense, 2020-08-02:
Hi again @matthewha123   Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-08-04 by matthewha123