realsenseai/librealsense #6874: Get distance info of all pixels immediately

Ted-EE, 2020-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |  Win10 64-bit | 
| Kernel Version (Linux Only)    |  N/A                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.0 }                          | 
| Language                            | python    | 
| Segment			| others                  | 

### Issue Description
Hi, I just met the same problem as https://github.com/IntelRealSense/librealsense/issues/2876#issuecomment-593446452. What I need to do is to remove the pixel whose distance is larger than 5 metres. However, it seems that using get_distance by traverse all pixels may take a bit long time and affect the performance of whole programme. Therefore, I would like to learn about if there exists a syntax which can show the distances of all pixels at once, so that I can remove the pixel whose distance is larger than 5 metres.

I have referred thelink Distance To Target (https://github.com/IntelRealSense/librealsense/issues/2876#issuecomment-596853831), but I don't think ROI will help me solve this problem, only a syntax to achieve distances of all pixels may help.

Thanks

MartyG-RealSense, 2020-07-21:
Hi @Ted-EE  If your goal is to exclude from the image all depth data that is greater than 5 meters then setting a **threshold** post-processing filter could accomplish that.  It enables you to set a minimum and maximum depth range, with data outside of that defined range being excluded from the image.

Example Python code for the filter is in the link below:

https://github.com/IntelRealSense/librealsense/issues/3002#issuecomment-473044109

Ted-EE, 2020-07-21:
Thanks for your replying. This syntax is nearly the answer, but what I express in the issue is unclear, I don't want to **remove** the pixel whose distance is larger than 5 m, I would like to **change** all the values of these pixels that are larger than 5 m to 255 instead of delete them.

MartyG-RealSense, 2020-07-21:
Before looking at changing all the pixels with your own code, it may be worth considering changing the "visual preset" of the colorizer to change how it colors the pixels.  This process is described on pages 13 and 14 of Intel's **Viewer User Guide**.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-Viewer-User-Guide.pdf

For example, setting the colorizer's visual preset to **Fixed** causes the image to be colored on a scale of the minimum distance (minz) up to a maximum of 6 meters.

[image]

The link below has Python scripting for setting the **color scheme** of the colorizer and the aforementioned colorizer **visual preset**.

https://github.com/IntelRealSense/librealsense/issues/3363

closed 2020-08-08 by MartyG-RealSense