realsenseai/librealsense #6875: L515 rgb frame failed align to infrared, need help.

cdb0y511, 2020-07-21:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.04.02) | 
| Operating System & Version |   (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.36.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 
Hi,I need to align the RGB frame to Infrared.

 pseudocode：
###################################################

        this->alignToColor = new rs2::align(RS2_STREAM_COLOR);
        this->alignToDepth = new rs2::align(RS2_STREAM_DEPTH);
        this->alignToInfrared = new rs2::align(RS2_STREAM_INFRARED);
        cfg.enable_stream(RS2_STREAM_DEPTH, 1024, 768, RS2_FORMAT_Z16, 30);// L515
        cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8,30);// L515
        cfg.enable_stream(RS2_STREAM_INFRARED, 1024, 768, RS2_FORMAT_Y8,30);// L515
        auto profile = pipe.start(cfg);
        auto frameSet = pipe.wait_for_frames(timeout_ms);
        auto infraredSet = alignToInfrared->process(frameSet) ;
            if (alignedColor) {
                auto rgb_stream_file = pipe.get_active_profile().get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
                frameSet = alignToColor->process(frameSet);
                width = rgb_stream_file.get_intrinsics().width;
                height = rgb_stream_file.get_intrinsics().height;
            }
            else {
                auto infrared_stream_file = pipe.get_active_profile().get_stream(RS2_STREAM_INFRARED).as<rs2::video_stream_profile>();
                width = infrared_stream_file.get_intrinsics().width;
                height = infrared_stream_file.get_intrinsics().height;
                frameSet = alignToDepth->process(frameSet);
                //frameSet = alignToInfrared->process(frameSet);
            }
       auto color = frameSet.get_color_frame();
       frames.rgbFrame_ = std::make_shared<cv::Mat>(Size(width, height), CV_8UC3, (void *) color.get_data(), Mat::AUTO_STEP);
####################################################
"frameSet = alignToDepth->process(frameSet);"  works fine, beside one problem that some point without depth estimation, the RGB frame corresponding point will be 0 too.
So I wish to use " frameSet = alignToInfrared->process(frameSet);", however, the RGB frame become totally wrong, like row and col misaligned. 
Thanks

RealSenseSupport, 2020-07-24:
Hi @cdb0y511 

I've been playing around with the align sample as my starting point:  https://github.com/IntelRealSense/librealsense/tree/master/examples/align as well as align-advanced https://github.com/IntelRealSense/librealsense/tree/master/examples/align-advanced

All as my starting point and haven't been able to see issues with aligning rgb to infrared.

cdb0y511, 2020-07-25:
> Hi @cdb0y511
>
> I've been playing around with the align sample as my starting point: https://github.com/IntelRealSense/librealsense/tree/master/examples/align as well as align-advanced [...]
Thanks, I will look into it. 
Maybe this is reason:
   frames.rgbFrame_ = std::make_shared<cv::Mat>(Size(width, height), CV_8UC3, (void *) color.get_data(), Mat::AUTO_STEP);
I wonder if there is a safe way to convert rs::video_frame to cv::Mat? 

PS, the align result is terrify , much more pixel misaligned than D400 series. I think I am not the only one to complain about the rgb to depth , depth to rgb alignment performance.  
The terrify align result may ruin the sensor in reconstruction usage. Any chance to improve it?

cdb0y511, 2020-07-25:
Hi@RealSenseSupport . @dorodnic
I try the align and align-advance.
I find the new rs2::align(RS2_STREAM_INFRARED) does not work at all. For both l515 and D400.
It means if you use
             if (dir == direction::to_depth)
        {
            // Align all frames to depth viewport
           // frameset = align_to_depth.process(frameset);
           frameset = align_to_infrared.process(frameset);
        }

The output rgb frame is original frame, which does not do any alignment. 
This explain why the 

  frames.rgbFrame_ = std::make_shared<cv::Mat>(Size(width, height), CV_8UC3, (void *) color.get_data(), Mat::AUTO_STEP);

looks so strange, width and height are original rgb width and height.
I change to below, it works. but the rgb frame does not align at all. This can be reproduce from D435i with 2.36.  pls fix it.
  
       frames.rgbFrame_ = std::make_shared<cv::Mat>(Size(color.get_width(), color.get_height()), CV_8UC3, (void *) color.get_data(), Mat::AUTO_STEP);

dorodnic, 2020-08-05:
Align to Infrared should be interpreted as align (depth) to infrared.
To align RGB to infrared, you need to align RGB to Depth. Depth and Infrared share the same view.
Currently, align supports two use cases - align depth to (stream) and align (stream) to depth. Alignment of (stream) to (stream) is not supported and not 100% well-defined, since depth is necessary for alignment.

cdb0y511, 2020-08-05:
> Align to Infrared should be interpreted as align (depth) to infrared.
> To align RGB to infrared, you need to align RGB to Depth. Depth and Infrared share the same view.
> Currently, align supports two [...]

Thanks for the reply,
However, there is still a reason that needs to align with infrared.
If align to depth, the pixel without depth estimation(or filtered, set to zero), RGB, or Infrared will be zero too.
I don't want a black hole in RGB or infrared images.
Any advice?

dorodnic, 2020-08-09:
Hi @cdb0y511 
The issue is that alignment is not a simple affine transform. You **need** depth to be able to do alignment. Pixel without depth can be mapped 5 pixels or 50 pixels, depending on if its 5 meters or 30 cm away. You could do alignment to depth after hole-filling post-processing, this may give acceptable results.

cdb0y511, 2020-08-09:
@dorodnic 
Thanks, this is the best solution for now.

closed 2020-08-09 by cdb0y511