realsenseai/librealsense #6876: SR305 incomplete bulk usb transfer

weseu, 2020-07-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | 2.32.1 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 | NVIDIA TX2 Jetpack4.2  |

I get athe follower massage.
$ roslaunch realsense2_camera rs_camera
[image]

and I try to fix it, but didn't solve the problem.
$./scripts/setup_udev_rules.sh

and I try to open realsense-view and set exposure ROI, I get the following massage.

Invalid Value in rs2_set_region_of_interest(sensor:0000019EF11ED820, min_x:805, min_y:388, max_x:983, max_y:519):
Incomplete bulk usb transfer!

[image]

MartyG-RealSense, 2020-07-21:
Hi @weseu  I was able to generate the **incomplete bulk usb transfer** error in the RealSense Viewer (SDK **2.35.2**) on every attempt with an original SR300 in a USB 3 port that is recognised by the Viewer as an SR305:

I followed the process below each time that I tested:

1.  Activate the RGB stream only
2.  Click the 'Set ROI' button
3.  Drag a ROI box on the RGB stream panel with the mouse of any size
4. The error pops up

The error occurred at every ROI box size, resolution and FPS tested.  

I then repeated the process on the latest **2.36.0** SDK and experienced exactly the same outcome.

Finally, I plugged the camera into my mains electricity powered USB 3 hub.  The error continued to occur.

As this problem can be replicated consistently and has been experienced by more than one person, I will file a bug report with Intel so that they can investigate.

weseu, 2020-07-22:
OK...if has any further news,please let me know~~  I need this function.

MartyG-RealSense, 2020-07-22:
I will definitely update this case when there is news to provide.

MartyG-RealSense, 2021-11-16:
Case closed due to the RealSense SR30x models being declared End of Life (retired).

closed 2021-11-16 by MartyG-RealSense