realsenseai/librealsense #6877: D435 realsense camera not detecting in virtual box ubuntu 18.04.

pp3993, 2020-07-21:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Linux|

Please provide a description of the problem

pp3993, 2020-07-21:
I am trying to install D435 realsense camera on my virtual ubutu

```

```

dorodnic, 2020-07-21:
Hi @pp3993 

https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-im-using-a-virtual-machine-and-the-camera-is-not-working

pp3993, 2020-07-21:
what should i try docker or VMware Workstation Player?

I tried with docker as well but same unable to detect. Can you help please with this .?

dorodnic, 2020-07-21:
"we recommend using VMware Workstation Player, and not Oracle VirtualBox for proper emulation of the USB3 controller"

I'll be happy to help with VMware since this is what worked for me, but not with the rest.

pp3993, 2020-07-21:
As you recommended I tried installing on vmware workstation 15 player. the device detected message coming after connected the cames but I am getting these errors while opening realsense viewer in terminal.
 [image]

this may be because of kernel `5.3.0-62-generic` as I have kernel series of 5 on my ubuntu 18.04 not kernel version (4.[4,8,10,13,15,16]])  installation guide.

also I have followed all the steps mentioned on given [link](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

 **can you please guide here what is wrong?**

[image]

dorodnic, 2020-07-21:
Hi @pp3993 
`dmesg` log and `lsusb` can help understand what is going on. Beforehand, please check in VM settings that USB3 is selected in USB controller settings. Also, please check that the D435 is connected to the VM in the top menu. 
Don't worry about the Kernel version at the moment, I don't see any issue related to it.

pp3993, 2020-07-27:
Thanks @dorodnic 
My camera working in VM ware now.

closed 2020-07-27 by pp3993