realsenseai/librealsense #6879: Unable to update firmware D415!

Dilip-Gurumurthy, 2020-07-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |  Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |             5.3.0                            | 
| Platform                                 | PC  |
| SDK Version                            |  2.29                     | 
| Language                            | Python 3                       | 
| Segment			|  Others                         | 

### Issue Description

I am unable to update FW!
I have tried using both rs-fw-update from command line:
`$ ./rs-fw-update -f Signed_Image_UVC_5_12_6_0.bin
`
As well as using realsense viewer.

But both the times the device gets disconnected in the middle and the update fails. The following events take place in realsense viewer, when update is initiated.
1. Back-up camera flash memory.
2. Completed back-up and store in some location.
3. At 30% progress. "Requesting to switch to Recovery mode"
4. "Intel Realsense D415 Disconnected."
5. "ERROR: Recovery device did not connect in time!"

MartyG-RealSense, 2020-07-21:
Hi @Dilip-Gurumurthy   Are you performing the firmware update with the short official USB cable supplied with the camera, or with a longer USB cable chosen by yourself please?

Also, when using **rs-fw-update**, is the camera plugged into a USB 3 port?

Dilip-Gurumurthy, 2020-07-21:
Hi @MartyG-RealSense  I use Ethernet cable which RealSense viewer detects as USB 3.2. Since I use a device(Lipsedge AE400) that is coupled with third-party software, I don't think standard methods of update will work. I might have to discuss this issue with the vendor itself. Sorry for posting here.

closed 2020-07-21 by Dilip-Gurumurthy

MartyG-RealSense, 2020-07-21:
No problem @Dilip-Gurumurthy   Yes, camera manufacturers such as LIPS Corp and FRAMOS are responsible for providing technical support for their RealSense cameras as they are not Intel products.  So support should be sought directly from them.  Good luck!