realsenseai/librealsense #6881: Vibration sensivity of L515

Leo16112, 2020-07-21:
Hello,

we just got your new L515 and are thinking about to place it on robot, that moves quite quickly(1 m/s) and stops suddenly. 
Are there any reports of testing of realsense L515 under vibration/kinetic shocks? We could find only this:

Joel Hagberg, Head of Product Management and Marketing at Intel RealSense. “Additionally, the MEMS mirror was designed to be resistant to shock, temperature fluctuations, and vibration.”

Also during our tests we could see, that camera output is influenced by its temperature. But we found a way to handle this. 

Kind Regards,
Anton.

RealSenseSupport, 2020-07-24:
Hi @Leo16112 

Are you specifically looking for information on the IMU that is being used on the L515 or the vibration information of the entire L515?  IMU information can be found in the datasheet:  https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet.  Also with regards to the IMU itself, you can do a calibration process that is outlined here:  https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

Unfortunately we cannot provide the details around our testing coverage of the camera.

Leo16112, 2020-07-24:
@RealSenseSupport Thank you for your answer. 
It is about entire L515. We are thinking to place the camera on the gripper of the industrial robot and we need to know shock and vibration limitations of the camera, so we could know if it is possible. The speed of the movement is similar to what you can see in this video.
https://youtu.be/TOl0ZPexRhI?t=33
If it possible to provide this information, it would be very nice.
Kind Regards,
Anton.

closed 2020-07-24 by Leo16112